tinygo-org/tinygo #1147: nRF53840-MDK Won’t Flash with new UF2 Bootloader

APDevice, 2020-05-31:
MakerDiary has replaced the Open Bootloader with the UF2 Bootloader. This appears to have broken existing support with the flash command.

APDevice, 2020-06-02:
The volume name is “MDK-DONGLE”. 

I think I managed to set up the file right, though I apparently can’t test it as I’ve now I’ve run into an issue with installing from source.  (opened a separate issue for that)

Raqbit, 2021-02-21:
For me the volume comes up as `MDK-DONGLE`.

My `INFO_UF2.TXT`: 

```
UF2 Bootloader 0.2.13-42-g82464f9-dirty lib/nrfx (v2.0.0) lib/tinyusb (legacy-1500-g23df777b) s140 6.1.1
Model: nRF52840 MDK USB Dongle
Board-ID: nRF52840-Dongle-v1
Date: Feb  3 2020
```

deadprogram, 2021-02-21:
Based on your description, the nrf52840-mdk.json should be changed to the following:

```json
{
    "inherits": ["nrf52840"],
    "build-tags": ["nrf52840_mdk"],
    "flash-1200-bps-reset": "true",
    "flash-method": "msd",
    "msd-volume-name": "MDK-DONGLE",
    "msd-firmware-name": "firmware.uf2",
    "uf2-family-id": "0xADA52840"
}
```

Can someone who has one of these boards try this out with the latest TinyGo `dev` branch?

Raqbit, 2021-02-21:
Now that I look at it, it does not look like the non-dongle [nrf52840-mdk](https://github.com/makerdiary/nrf52840-mdk) has the same bootloader as the [nrf52840-mdk-usb-dongle](https://github.com/makerdiary/nrf52840-mdk-usb-dongle). In fact, despite sharing a name these boards appear to be quite different, as the non-dongle version has DAPlink running on a separate MCU where the dongle-version used to be shipped with Nordic's Open Bootloader, which has been replaced with the UF2 bootloader on february 4th of 2020.

Even though Makerdiary links to `machine/nrf52840-mdk` for tinygo support on the [dongle-variant wiki](https://wiki.makerdiary.com/nrf52840-mdk-usb-dongle/), I am not sure if this was actually tested and confirmed working.

deadprogram, 2021-02-21:
I see. So actually adding a new board definition in TinyGo for the dongle is probably the correct move, since it also has a different pinout.

deadprogram, 2021-03-08:
This has been released with v0.17.0 so now closing. Thanks!

PR #1654 (2021-02-24): Add nrf52840-mdk-usb-dongle target
Adds support for the [nrf52840-mdk-usb-dongle](https://wiki.makerdiary.com/nrf52840-mdk-usb-dongle/), with the now-default UF2 bootloader. 

This target also expects the S140 v6 SoftDevice to be present, which might not be the case if one of the non-bluetooth C++ examples, like the [C++ blinky example](https://github.com/makerdiary/nrf52840-mdk-usb-dongle/tree/cbb51081997ce31fdf355bd533cad5f5d575f347/examples/nrf5-sdk/blinky) has been flashed (see [linker script](https://github.com/makerdiary/nrf52840-mdk-usb-dongle/blob/cbb51081997ce31fdf355bd533cad5f5d575f3 [... cut]
closed 2021-03-08 by deadprogram