tinygo-org/tinygo #1148: cgo: Looks buggy when in common library

racerxdl, 2020-06-01:


aykevl, 2020-06-08:
Probably not the issue, but I see that the `hue` function is a variadic function call (which is easy to miss in C). You might want to specify this instead:

    void hue(void) {}

Also, function definitions are currently not compiled in CGo headers.

racerxdl, 2020-06-08:
To be honest, thats totally optional to simulate the issue. I just put that to use C, but this also gives the same error:

```go
import (
    _ "unsafe"
)
/*
#cgo LDFLAGS: -lnx
*/
import "C"
```

Just importing C is enough to cause the error. The code I provided was just to use something from Cgo.

aykevl, 2020-06-23:
I can't reproduce this. Can you maybe make a small repository or gist with all the files necessary for this?

I made two files:

```go
// cgo-import-test/cgo.go
package importtest

import (
	_ "unsafe"
)

/*
#cgo LDFLAGS: -lnx
*/
import "C"

// testfile.go
package main

import _ "./cgo-import-test"

func main() {}
```

I get the following error:

```
$ tinygo run testfile.go
ld: error: unable to find library -lnx
collect2: error: ld returned 1 exit status
error: failed to link /tmp/tinygo421895540/main: exit status 1
```

This error is entirely expected.

Additionally, I'm sure CGo usually works when used in packages as I frequently use it for the github.com/tinygo-org/bluetooth package.

racerxdl, 2020-07-08:
@aykevl well that does not happen doing import from different package of the same project. It needs to be a external library, and I noticed that it needs to use go-modules. For example:
https://github.com/racerxdl/gonx/blob/dev/base.go#L6-L7

Just trying to import that `gonx` breaks the compilation with the message I said.

racerxdl, 2020-07-10:
Ok, I just found it was some cflags defined for the nintendo switch that was breaking the compilation. It's not related to tinygo!

Sorry for the mess :((

closed 2020-07-10 by racerxdl

aykevl, 2020-07-12:
Glad you found the issue! I was surprised, as I had been using CGo (with and without Go modules) for a while for BLE support.

racerxdl, 2020-07-12:
Yeah, I was trying to narrow down the issues, then I found that if the target wasn't nintendoswitch it worked perfectly. Solved now :D