tinygo-org/tinygo #1150: flash ignores all but first source filename

AgentZombie, 2020-06-01:
`version: 0.13.1`

Given two source files:

a.go:

```
package main

import "machine"

func main() {
	p := machine.Pin(5)
	p.Configure(machine.PinConfig{
		Mode: machine.PinOutput,
	})
	for {
		blink(p)
	}
}
```

b.go:

```
package main

import (
	"machine"
	"time"
)

func blink(p machine.Pin) {
	p.High()
	time.Sleep(500 * time.Millisecond)
	p.Low()
	time.Sleep(500 * time.Millisecond)
}
```

Trying to compile by specifically listing the filenames results in only the first file being consulted:

```
$ tinygo flash -target arduino a.go b.go
# a.go
a.go:11:3: undeclared name: blink
```

```
$ tinygo flash -target arduino b.go a.go
error: function main is undeclared in the main package
```

Expected behavior: All source files listed on the command line are read.

deadprogram, 2020-06-07:
I updated the description.

deadprogram, 2024-12-13:
This is still an issue from what I can tell.

aykevl, 2024-12-14:
Yes, did a quick test. Indeed still an issue.

davecheney, 2026-10-28:
Upstream `go run` accepts multiple `.go` files from the same package and compiles them together. #3150 tracks the same limitation for `tinygo run`, where additional filenames are currently treated as program arguments. This issue tracks the broader need for multi-file input across `build`, `flash`, and `run`.