tinygo-org/tinygo #1151: Getting "fatal error: 'llvm-c/Analysis.h’ " when compiling from source

APDevice, 2020-06-02:
I ran into this error while attempting to do a source install:

`$ go install
go: downloading tinygo.org/x/go-llvm v0.0.0-20200401165421-8d120882fc7a
go: downloading go.bug.st/serial v1.0.0
go: downloading github.com/blakesmith/ar v0.0.0-20150311145944-8bd4349a67f2
go: downloading golang.org/x/tools v0.0.0-20200216192241-b320d3a0f5a2
go: downloading github.com/google/shlex v0.0.0-20181106134648-c34317bd91bf
go: downloading github.com/marcinbor85/gohex v0.0.0-20180128172054-7a43cd876e46
go: downloading golang.org/x/sys v0.0.0-20191128015809-6d18c012aee9
go: downloading github.com/creack/goselect v0.1.1
# tinygo.org/x/go-llvm
../../../go/pkg/mod/tinygo.org/x/go-llvm@v0.0.0-20200401165421-8d120882fc7a/analysis.go:16:10: fatal error: 'llvm-c/Analysis.h' file not found
#include "llvm-c/Analysis.h" // If you are getting an error here read bindings/go/README.txt
         ^~~~~~~~~~~~~~~~~~~
1 error generated.
`
I did follow up with the readme, and got this:
`$ $GOPATH/src/llvm.org/llvm/bindings/go/build.sh
-bash: /src/llvm.org/llvm/bindings/go/build.sh: No such file or directory`

go version: go1.14.3 darwin/amd64
llvm version: 10.0.0_3
macOS version: 10.14.6 (Mojave)

aykevl, 2020-06-04:
I have updated the formatting a bit. Hint: you can use three backticks to format code, like this:

```
`​``
some code
`​``
```

To fix your issue, have you installed LLVM 10 using Homebrew? The command is this:

    brew install llvm

The comment in the error is misleading. The problem is that the LLVM installation could not be found.

APDevice, 2020-06-05:
I tried that. It told me it was already installed. I even tried the reinstall command that it offered. No luck.

```
Warning: llvm 10.0.0_3 is already installed and up-to-date
To reinstall 10.0.0_3, run `brew reinstall llvm`
$ brew reinstall llvm
==> Downloading https://homebrew.bintray.com/bottles/llvm-10.0.0_3.mojave.bottle
Already downloaded: /Users/dylanluttrell/Library/Caches/Homebrew/downloads/6d4c3816f98949b64550d4a36656b2661f8e5aeea36a90abbdbea68c8215b9a2--llvm-10.0.0_3.mojave.bottle.tar.gz
==> Reinstalling llvm 
==> Pouring llvm-10.0.0_3.mojave.bottle.tar.gz
==> Caveats
To use the bundled libc++ please add the following LDFLAGS:
  LDFLAGS="-L/usr/local/opt/llvm/lib -Wl,-rpath,/usr/local/opt/llvm/lib"

llvm is keg-only, which means it was not symlinked into /usr/local,
because macOS already provides this software and installing another version in
parallel can cause all kinds of trouble.

If you need to have llvm first in your PATH run:
  echo 'export PATH="/usr/local/opt/llvm/bin:$PATH"' >> /Users/dylanluttrell/.bash_profile

For compilers to find llvm you may need to set:
  export LDFLAGS="-L/usr/local/opt/llvm/lib"
  export CPPFLAGS="-I/usr/local/opt/llvm/include”
```

aykevl, 2020-06-06:
I think I've seen this issue before, I'm not sure why it's happening (as it appears to work in many cases).

Can you share the output of the following command?

```
ls -ld /usr/local/opt/llvm*
```

aykevl, 2020-06-06:
I think you can fix this by specifically installing LLVM 10:

    brew install llvm@10

APDevice, 2020-06-10:
Sorry for the delay. When I run ls -ld /usr/local/opt/llvm* I get this:
```
lrwxr-xr-x  1 <username>  admin  23 Jun  5 03:18 /usr/local/opt/llvm -> ../Cellar/llvm/10.0.0_3
lrwxr-xr-x  1 <username>  admin  23 Jun  5 03:18 /usr/local/opt/llvm@10 -> ../Cellar/llvm/10.0.0_3
```

I tried the brew install, but I’m still getting that error.

aykevl, 2020-06-17:
Weird. The paths look correct. Can you check whether `/usr/local/opt/llvm@10/include/llvm-c/Analysis.h` exists, and if not what the first parent is that does?
What might help is uninstalling and reinstalling LLVM.

campoy, 2020-06-21:
In case it helps, I had the same issue and I was able to go over this first hurdle by setting `export C_INCLUDE_PATH=/usr/local/Cellar/llvm/10.0.0_3/include`

Unfortunately, that only took me to a larger error that I'm including below:

<details><code><pre>
# tinygo.org/x/go-llvm
In file included from IRBindings.cpp:13:
In file included from ../../../../pkg/mod/tinygo.org/x/go-llvm@v0.0.0-20200401165421-8d120882fc7a/IRBindings.h:19:
In file included from /usr/local/Cellar/llvm/10.0.0_3/include/llvm/IR/Metadata.h:18:
In file included from /usr/local/Cellar/llvm/10.0.0_3/include/llvm/ADT/ArrayRef.h:12:
/usr/local/Cellar/llvm/10.0.0_3/include/llvm/ADT/Hashing.h:266:16: error: no matching constructor for initialization of 'llvm::hashing::detail::hash_state'
/usr/local/Cellar/llvm/10.0.0_3/include/llvm/ADT/Hashing.h:259:8: note: candidate constructor (the implicit copy constructor) not viable: requires 1 argument, but 7 were provided
/usr/local/Cellar/llvm/10.0.0_3/include/llvm/ADT/Hashing.h:259:8: note: candidate constructor (the implicit move constructor) not viable: requires 1 argument, but 7 were provided
/usr/local/Cellar/llvm/10.0.0_3/include/llvm/ADT/Hashing.h:259:8: note: candidate constructor (the implicit default constructor) not viable: requires 0 arguments, but 7 were provided
In file included from IRBindings.cpp:13:
In file included from ../../../../pkg/mod/tinygo.org/x/go-llvm@v0.0.0-20200401165421-8d120882fc7a/IRBindings.h:19:
In file included from /usr/local/Cellar/llvm/10.0.0_3/include/llvm/IR/Metadata.h:18:
In file included from /usr/local/Cellar/llvm/10.0.0_3/include/llvm/ADT/ArrayRef.h:15:
/usr/local/Cellar/llvm/10.0.0_3/include/llvm/ADT/STLExtras.h:559:49: error: no template named 'index_sequence' in namespace 'std'
/usr/local/Cellar/llvm/10.0.0_3/include/llvm/ADT/STLExtras.h:564:36: error: no template named 'index_sequence' in namespace 'std'
/usr/local/Cellar/llvm/10.0.0_3/include/llvm/ADT/STLExtras.h:569:36: error: no template named 'index_sequence' in namespace 'std'
/usr/local/Cellar/llvm/10.0.0_3/include/llvm/ADT/STLExtras.h:576:46: error: no member named 'index_sequence_for' in namespace 'std'
/usr/local/Cellar/llvm/10.0.0_3/include/llvm/ADT/STLExtras.h:576:65: error: 'Iters' does not refer to a value
/usr/local/Cellar/llvm/10.0.0_3/include/llvm/ADT/STLExtras.h:551:41: note: declared here
/usr/local/Cellar/llvm/10.0.0_3/include/llvm/ADT/STLExtras.h:579:23: error: no member named 'index_sequence_for' in namespace 'std'
/usr/local/Cellar/llvm/10.0.0_3/include/llvm/ADT/STLExtras.h:579:42: error: 'Iters' does not refer to a value
/usr/local/Cellar/llvm/10.0.0_3/include/llvm/ADT/STLExtras.h:551:41: note: declared here
/usr/local/Cellar/llvm/10.0.0_3/include/llvm/ADT/STLExtras.h:583:30: error: no member named 'index_sequence_for' in namespace 'std'
/usr/local/Cellar/llvm/10.0.0_3/include/llvm/ADT/STLExtras.h:583:49: error: 'Iters' does not refer to a value
/usr/local/Cellar/llvm/10.0.0_3/include/llvm/ADT/STLExtras.h:551:41: note: declared here
/usr/local/Cellar/llvm/10.0.0_3/include/llvm/ADT/STLExtras.h:590:30: error: no member named 'index_sequence_for' in namespace 'std'
/usr/local/Cellar/llvm/10.0.0_3/include/llvm/ADT/STLExtras.h:590:49: error: 'Iters' does not refer to a value
/usr/local/Cellar/llvm/10.0.0_3/include/llvm/ADT/STLExtras.h:551:41: note: declared here
/usr/local/Cellar/llvm/10.0.0_3/include/llvm/ADT/STLExtras.h:610:18: error: no template named 'index_sequence' in namespace 'std'
/usr/local/Cellar/llvm/10.0.0_3/include/llvm/ADT/STLExtras.h:622:30: error: no member named 'index_sequence_for' in namespace 'std'
/usr/local/Cellar/llvm/10.0.0_3/include/llvm/ADT/STLExtras.h:622:49: error: 'Iters' does not refer to a value
/usr/local/Cellar/llvm/10.0.0_3/include/llvm/ADT/STLExtras.h:606:23: note: declared here
/usr/local/Cellar/llvm/10.0.0_3/include/llvm/ADT/STLExtras.h:639:28: error: no template named 'index_sequence' in namespace 'std'
/usr/local/Cellar/llvm/10.0.0_3/include/llvm/ADT/STLExtras.h:642:50: error: no template named 'index_sequence' in namespace 'std'
/usr/local/Cellar/llvm/10.0.0_3/include/llvm/ADT/STLExtras.h:650:28: error: no member named 'index_sequence_for' in namespace 'std'
/usr/local/Cellar/llvm/10.0.0_3/include/llvm/ADT/STLExtras.h:650:47: error: 'Args' does not refer to a value
/usr/local/Cellar/llvm/10.0.0_3/include/ll# tinygo.org/x/go-llvm
In file included from IRBindings.cpp:13:
In file included from ../../../../pkg/mod/tinygo.org/x/go-llvm@v0.0.0-20200401165421-8d120882fc7a/IRBindings.h:19:
In file included from /usr/local/Cellar/llvm/10.0.0_3/include/llvm/IR/Metadata.h:18:
In file included from /usr/local/Cellar/llvm/10.0.0_3/include/llvm/ADT/ArrayRef.h:12:
/usr/local/Cellar/llvm/10.0.0_3/include/llvm/ADT/Hashing.h:266:16: error: no matching constructor for initialization of 'llvm::hashing::detail::hash_state'
/usr/local/Cellar/llvm/10.0.0_3/include/llvm/ADT/Hashing.h:259:8: note: candidate constructor (the implicit copy constructor) not viable: requires 1 argument, but 7 were provided
/usr/local/Cellar/llvm/10.0.0_3/include/llvm/ADT/Hashing.h:259:8: note: candidate constructor (the implicit move constructor) not viable: requires 1 argument, but 7 were provided
/usr/local/Cellar/llvm/10.0.0_3/include/llvm/ADT/Hashing.h:259:8: note: candidate constructor (the implicit default constructor) not viable: requires 0 arguments, but 7 were provided
In file included from IRBindings.cpp:13:
[... cut ...]
/usr/local/Cellar/llvm/10.0.0_3/include/llvm/ADT/STLExtras.h:650:28: error: no member named 'index_sequence_for' in namespace 'std'
/usr/local/Cellar/llvm/10.0.0_3/include/llvm/ADT/STLExtras.h:650:47: error: 'Args' does not refer to a value
/usr/local/Cellar/llvm/10.0.0_3/include/llvm/ADT/STLExtras.h:626:60: note: declared here
fatal error: too many errors emitted, stopping now [-ferror-limit=]
vm/ADT/STLExtras.h:626:60: note: declared here
fatal error: too many errors emitted, stopping now [-ferror-limit=]
</pre></code></details>

deadprogram, 2020-06-22:
I will boot up my test macOS system and give this a try myself. 

So to be clear, what you are wanting to do is a source install of TinyGo using your homebrew install of llvm 20, correct?

aykevl, 2020-06-23:
I should add that I cannot reproduce this issue locally and that it works in CI. Therefore I suspect there is something wrong with the LLVM installation.

deadprogram, 2024-11-27:
Closing due to age. Thank you!