tinygo-org/tinygo #1156: Fix the git log

sago35, 2020-06-03:
The git log I was writing was bad, so the dev branch has a bad log.
I think we can ammend it now, what do you think?

```
64d51b2 Extend SAMD51 pinPadMapping
f103e91 Add SAMD51 pin change interrupt settings
40afeea Add SAMD51 ADC settings
9712297 Add SAMD51 pins
72064e1 WIP flash: fix touchSerialPortAt1200bps on windows
```

deadprogram, 2020-06-04:
> the dev branch has a bad log.

Sorry, I do not understand. What do you mean exactly by this?

sago35, 2020-06-04:
Other commits have a prefix such as `sam:`, so a similar log is better.
I didn't follow the rules for my commit because I thought it would be a squash merge.

And especially the character `WIP` is not good.
I think this is because I named it WIP at the time I created the PR.

before
```
64d51b2 Extend SAMD51 pinPadMapping
f103e91 Add SAMD51 pin change interrupt settings
40afeea Add SAMD51 ADC settings
9712297 Add SAMD51 pins
72064e1 WIP flash: fix touchSerialPortAt1200bps on windows
```

after
```
sam: extend samd51 pinPadMapping
sam: add samd51 pin change interupt settings
sam: add samd51 adc setting
sam: add samd51 pins
flash: fix touchSerialPortAt1200bps on windows
```

sago35, 2020-06-04:
If you want to fix it, you'll need a force push.
I think there is an option of not fixing it.

deadprogram, 2020-06-04:
Oh, I see. I should have noticed that, my bad. 

Probably not important enough to force push to `dev` which will then likely cause some of the current unmerged branches some extra work.

I will keep an eye open for that next time.

aykevl, 2020-06-04:
The main purpose of the prefix is to make it slightly easier to check the git log. The prefix allows quickly scanning for the part of the code you're interested in, and most larger open source projects use it. While unfortunate, I don't think rewriting history is a good idea for such a trivial thing (especially after so much time has passed).

Mistakes happen, but I don't think it's a problem.

closed 2020-06-04 by aykevl