tinygo-org/tinygo #1161: Benchmarks

pjebs, 2020-06-06:
Since there is no GC, I was wondering how fast tiny go is compared to standard go, C and rust for various "standard" math algorithms they typically use for benchmarks.

aykevl, 2020-06-06:
There definitely is a GC, and it is enabled by default.

I haven't really done much speed benchmarking as the primary goal is small programs. I did a benchmark once, I believe [n-body](https://benchmarksgame-team.pages.debian.net/benchmarksgame/performance/nbody.html), and at the time the performance between standard Go and TinyGo was near identical.

Feel free to do some benchmarking yourself. But remember that it heavily depends on the specific programs and in some cases it may not matter much. Specifically for math-heavy algorithms, it's unlikely the difference will be big between Clang and TinyGo (comparing against Clang or Rust would be fairer, as it uses the same backend). And for math-heavy algorithms, it's unlikely the GC will be relevant at all.
If you do benchmarking, add the `-opt=2` flag to optimize for speed over size (the default is to optimize for size, and I've seen a case where it made a 10% or so difference).

deadprogram, 2020-06-07:
It would be interesting to have some TinyGo benchmarks, but I personally am more interested in measuring against ourselves over time to have some quantitative info about performance improving or degrading.

pjebs, 2020-06-07:
@deadprogram On this site: https://benchmarksgame-team.pages.debian.net/benchmarksgame/fastest/rust-go.html you can see various programming languages benchmarked against each other for various algorithms. When you click on the algorithm, you can see the Go code.

pjebs, 2020-06-07:
@aykevl For AVR documentation says no GC. Is the name of the game when dealing with tinygo do ensure nothing ends up in the heap?

jaddr2line, 2020-06-07:
That documentation needs to be fixed then. We used to fail to compile when there was a memory allocation on AVR, but now we have GC on by default on AVR.

deadprogram, 2023-05-23:
Closing due to age. Please reopen if needed, thanks!

closed 2023-05-23 by deadprogram