tinygo-org/tinygo #1166: I2C hangs on Arduino Uno

Arianna-Ko1in, 2020-06-08:
The sequence of two I2C requests leads to infinite waiting of the response, although the same sequence, written in C++, works correctly. The test was made on the ADXL345 chip, connected via I2C to Arduino UNO (pins A4-A5).

Steps to reproduce:
```
i2c := machine.I2C0
i2c.Configure(machine.I2CConfig{})
buf = []byte{0,0}
i2c.Tx(ADXL345, []byte{0x32}, buf)
//next command leads to infinite loop
i2c.Tx(ADXL345, []byte{0x36}, buf)
```
We've found a workaround for this issue by sending only write transmission with the same register and nil in the read buffer (i.e. 
```
i2c.Tx(ADXL345, []byte{0x32}, nil) 
```
between commands.

deadprogram, 2020-06-17:
Confirmed this issue. The `Connected()` returns valid, but after that the device becomes non-responsive. Also confirmed this is the case with any I2C device.

deadprogram, 2020-06-25:
Further info: after some testing seems `master` branch works, but `dev` branch was able to reproduce. I then started in on a `git bisect` session. Here is what I found.

In the following commit, numeric values output to the serial port are garbage:

```
$ git bisect bad
fc4857e98c1c315a92ed687bc45ff895f0eb9600 is the first bad commit
commit fc4857e98c1c315a92ed687bc45ff895f0eb9600
Author: Ayke van Laethem <aykevanlaethem@gmail.com>
Date:   Sat Apr 4 14:58:56 2020 +0200

    runtime: avoid recursion in printuint64 function
    
    This function is called from runtime.printitf, which is called from
    runtime._panic, and is therefore the leaf function of many call paths.
    This makes analyzing stack usage very difficult.
    
    Also forwarding printuint32 to printuint64 as it reduces code size in
    the few examples I've tested. Printing numbers is not often done so it
    doesn't matter if it's a bit slow (the serial connection is probably
    slower anyway).
```

Here is an example of the expected output:

```
Terminal ready                                                                                                                                                               
32714 -3417 979736                                                                                                                                                           
33935 1708 967529                                                                                                                                                            
28808 -3417 979492                                                                                                                                                           
33203 -5859 982421                                                                                                                                                           
30273 -10742 964843                                                                                                                                                          
35156 -3173 976562                                                                                                                                                           
33691 244 985107                                                                                                                                                             
33691 0 972656
```

Invalid output:

```
Terminal ready                                                                                                                                                               
qqqqqqqqqqqqqqqqqqqz qqqqqqqqqqqqqqqqqqqz qqqqqqqqqqqqqqqqqqqz                                                                                                               

```

@aykevl what do you think?

aykevl, 2020-06-30:
Thank you for bisecting this. Yes, that's clearly a bug and we should fix it before the next release.

aykevl, 2020-06-30:
Actually, I think this is an AVR backend bug. When I try it out with my local LLVM 11 branch, the following code works correctly while using LLVM 10 prints garbage for numbers above 10:

```go
package main

func main() {
	for i := 1; i < 500; i *= 3 {
		println(i)
	}
}
```

Not sure what to do. Perhaps we can roll back then change just for AVR as a workaround?

aykevl, 2020-06-30:
See https://github.com/tinygo-org/tinygo/pull/1202

deadprogram, 2020-06-30:
Seems like fixed to me using #1202 with MPU-6050 on Arduino Uno.

Marking for next release.

deadprogram, 2020-08-07:
This was released with v0.14.0 so now closing. Please reopen if needed. Thanks!

closed 2020-08-07 by deadprogram