tinygo-org/tinygo #1184: wasm Can't return strng

milkbobo, 2020-06-23:
`
package main

import (
	"crypto/sha256"
	"encoding/hex"
	"strconv"
)

func main() {
        println("multiple two numbers:", add(3,2)) // expecting 5
}

// This function is exported to JavaScript, so can be called using
// exports.add() in JavaScript.
//go:export add
func add(timestamp,x int) string {
   	str:= strconv.Itoa(timestamp)+"hongbeibangxiaobaixiaoliang520"

	//使用sha256哈希函数
	h := sha256.New()
	h.Write([]byte(str))
	sum := h.Sum(nil)
    println(timestamp)
    println(x)
    // println(hex.EncodeToString(sum))
    aa := hex.EncodeToString(sum)
    println(aa)
    return aa
}
`
Run the command : tinygo build -o wasm.wasm -target wasm ./main.go

In the console is running :
[image]

aykevl, 2020-06-23:
Returning a string is not supported by WebAssembly. Only integers (`int`, `int64`, etc) and float (`float32`, `float64`) are supported.
I do not think TinyGo can do a lot about this.

sepbot, 2020-10-05:
Is it somehow possible to use `WebAssembly.Memory` instead to share data?https://github.com/WebAssembly/design/blob/master/JS.md#webassemblymemory-objects

codefromthecrypt, 2022-04-21:
Here is an example that compiles fine in TinyGo and works in wazero. It shows one pattern for moving strings in and out. This will be merged to https://github.com/tetratelabs/wazero/tree/main/examples soon

```go
package main

import (
	"fmt"
	"reflect"
	"unsafe"
)

// main is required for TinyGo to compile to Wasm.
func main() {}

/// greet prints a greeting to the console.
func greet(name string) {
	log(fmt.Sprint("wasm >> ", greeting(name)))
}

/// Logs a message to the console using _log.
func log(message string) {
	buf := []byte(message)
	ptr := &buf[0]
	unsafePtr := uintptr(unsafe.Pointer(ptr))
	_log(uint32(unsafePtr), uint32(len(buf)))
}

// _log is a WebAssembly import which prints a string (linear memory offset,
// byteCount) to the console.
//
// Note: In TinyGo "//export" on a func is actually an import!
//go:wasm-module env
//export log
func _log(ptr uint32, size uint32)

// greeting gets a greeting for the name.
func greeting(name string) string {
	return fmt.Sprint("Hello, ", name, "!")
}

// _greet is a WebAssembly export that accepts a string pointer (linear
// memory offset) and calls greet.
//export greet
func _greet(ptr, size uint32) {
	name := ptrToString(ptr, size)
	greet(name)
}

// _greet is a WebAssembly export that accepts a string pointer (linear
// memory offset) and returns a pointer/size pair packed into a uint64.
//
// Note: This uses a uint64 instead of two result values for compatibility
// with WebAssembly 1.0.
//export greeting
func _greeting(ptr, size uint32) (ptrSize uint64) {
	name := ptrToString(ptr, size)
	g := greeting(name)
	ptr, size = stringToPtr(g)
	return (uint64(ptr) << uint64(32)) | uint64(size)
}

// ptrToString returns a string from WebAssembly compatible numeric
// types representing its pointer and length.
func ptrToString(ptr uint32, size uint32) (ret string) {
	// Here, we want to get a string represented by the ptr and size. If we
	// wanted a []byte, we'd use reflect.SliceHeader instead.
	strHdr := (*reflect.StringHeader)(unsafe.Pointer(&ret))
	strHdr.Data = uintptr(ptr)
	strHdr.Len = uintptr(size)
	return
}

// stringToPtr returns a pointer and size pair for the given string
// in a way that is compatible with WebAssembly numeric types.
func stringToPtr(s string) (uint32, uint32) {
	buf := []byte(s)
	ptr := &buf[0]
	unsafePtr := uintptr(unsafe.Pointer(ptr))
	return uint32(unsafePtr), uint32(len(buf))
}
```

codefromthecrypt, 2022-04-21:
note: the above hasn't anything to do with the javascript type mapping, which is important if the goal is to be used in a browser.

xmsz, 2022-05-11:
any solution?

deadprogram, 2024-11-30:
Closing since more info on this is covered in many places. Thanks!

closed 2024-11-30 by deadprogram