tinygo-org/tinygo #1186: build Linux arm error!!

xywf221, 2020-06-24:
Command parameters : `tinygo `build` -o test_cli -target arm-32-linux ./test.go`
error:`
/usr/lib/gcc-cross/arm-linux-gnueabihf/9/../../../../arm-linux-gnueabihf/bin/ld: error: /tmp/tinygo986353092/main uses VFP register arguments, /tmp/tinygo986353092/main.o does not
/usr/lib/gcc-cross/arm-linux-gnueabihf/9/../../../../arm-linux-gnueabihf/bin/ld: failed to merge target specific data of file /tmp/tinygo986353092/main.o
collect2: error: ld returned 1 exit status
error: failed to link /tmp/tinygo986353092/main: exit status 1
`
linux version : wsl ubuntu 20.04
clang version 10.0.0-4ubuntu1
gcc version 9.3.0 (Ubuntu 9.3.0-10ubuntu2)
test.go:```go
package main

func main(){
        println("tinygo-arm-testing")
}
```

jaddr2line, 2020-06-24:
That is not a valid LLVM target spec - `32` is not a vendor or system. I think what you are looking for is probably `arm-linux-gnueabihf`.

https://clang.llvm.org/docs/CrossCompilation.html

aykevl, 2020-06-24:
If you want to cross compile from amd64 Linux to arm Linux, all you have to do is set `GOARCH` (just like normal Go).

deadprogram, 2020-07-06:
We should add some info on this here https://github.com/tinygo-org/tinygo-site/blob/dev/content/linux/_index.md

xywf221, 2020-07-07:
I don't think I'll try tinygo again in a short time. I've used rust to develop applications!

closed 2020-07-07 by xywf221

aykevl, 2020-07-07:
I have written some documentation how to do this:

https://github.com/tinygo-org/tinygo-site/pull/95