tinygo-org/tinygo #1187: ___tinygo_gpio_set Error

MisterLight, 2020-06-24:
Hi everybody, Im trying to build the blinky1 example by myself, so I copy the blinky.go code from example folder and trying to build it, like below :
tinygo build -o ./build/main  main.go

The code :

```
package main

import (
	"machine"
	"time"
)

// start here at main function
func main() {

	led := machine.LED
	led.Configure(machine.PinConfig{Mode: machine.PinOutput})
	for {
		led.Low()
		time.Sleep(time.Millisecond * 500)

		led.High()
		time.Sleep(time.Millisecond * 500)
	}
}
```

But its give me the following error :

LED not declared by package machine

So I looked at machine package  and change machine.LED for machine.Pin(13)

```
        led := machine.LED
	led.Configure(machine.PinConfig{Mode: machine.PinOutput})
	for {
		led.Low()
		time.Sleep(time.Millisecond * 500)

		led.High()
		time.Sleep(time.Millisecond * 500)
	}
```

I build it and give me the following error:

```
Undefined symbols for architecture x86_64:
  "___tinygo_gpio_set", referenced from:
      _main.go.main in main.o
      _main.go.main.resume in main.o
  "___tinygo_gpio_configure", referenced from:
      _(machine.Pin).Configure in main.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
error: failed to link /var/folders/qd/j7gjsfg90n300l2jfvx9m4m00000gn/T/tinygo994926948/main: exit status 1
```

Im using Mac, Im just download the binaries using homebrew and setting up my GOPATH to use tinygo folder
 GOPATH="/Users/jeiie/go:/usr/local/Cellar/tinygo/0.13.1"

Can anyone help me?

jaddr2line, 2020-06-24:
You cannot run blinky on a mac, as there are no GPIO pins on a mac. You need to specify a target for the machine you are targeting. For example, for an Arduino Uno:
```
tinygo build -target arduino -o ./build/main main.go
```
and to upload instead of building:
```
tinygo flash -target arduino main.go
```

MisterLight, 2020-06-24:
Yeah, Its works!

its was the -target flag.

Ty so much! :)

closed 2020-06-24 by jaddr2line