tinygo-org/tinygo #1189: Ability to reduce js-centricness of compiled wasm modules

dsherret, 2020-06-25:
The compiled wasm modules are very JS-centric. Nowadays wasm is being used in areas other than in the web browser.

Would it be possible to have a compile flag that gets rid of all the `"syscall/js*"` imports and doesn't have a `fd_write` import? I feel like `fd_write` should only show up when targeting `wasm-wasi` similar to how Rust works (so doing a print would be a no-op).

jaddr2line, 2020-06-27:
I think we may want to split wasm into multiple subtargets. We need a target compatible with Go's wasm+js target, and also a WASI target (although we will want to [sync up with the standard Go implementation](https://github.com/golang/go/issues/31105)).

> I feel like fd_write should only show up when targeting wasm-wasi

Is there a specific platform you were thinking about? No matter what, the TinyGo runtime needs a few things in order to work:
1. a print output destination
2. a mechanism for checking the current time
3. a mechanism for setting timers

dsherret, 2020-06-27:
@jaddr2line why does it need to be able to print? Could printing be a no-op when fd_write doesn't exist? In Rust it's possible to compile to `wasm32-unknown-unknown` without any exports or imports (no need for printing).

I do not know enough about the go runtime or go in general, but is it possible to have code that doesn't check the current time or set timers? (I'm guessing the runtime uses it for garbage collection? If so, perhaps people could disable garbage collection then run it manually and not need those imports? I have no idea how it works...)

jaddr2line, 2020-06-27:
> is it possible to have code that doesn't check the current time or set timers?

The scheduler uses this for timers.

jaddr2line, 2020-06-27:
> Could printing be a no-op when fd_write doesn't exist?

Yes, but this effectively limits panics to showing you "something happened" with no description as to what the panic pertains to.

dsherret, 2020-06-28:
> Yes, but this effectively limits panics to showing you "something happened" with no description as to what the panic pertains to.

That is what Rust does and to me this is ok. When I call a wasm function export and it panics, I often only need to know the inputs in order to debug the problem (which I have on the calling side).

It would be nice if the compiler didn't force me to implement its specific WASM interface for outputting a message, even if that interface happens to align with WASI (people should be able to target `wasm-wasi` if they want that).

> The scheduler uses this for timers.

So if someone is not using any timers then I believe the exports are unnecessary? I don't think they need to be imposed on every module. If the code does use timers and the target is `wasm32-unknown-unknown` then perhaps it could panic when compiling or at runtime when a timer is run.

I think Rust has got this right in that targeting `wasm32-unknown-unknown` does not have any imports or exports by default—it doesn't force a specific interface on the WASM module by the compiler and doesn't enforce having a `main` function. This makes it very easy to define a WASM module interface and then write Rust WASM modules that implement that interface. At the current moment, this is not possible compiling with Go or TinyGo because the code that runs the WASM modules implementing a custom interface needs to have special handling for Go modules.

aykevl, 2020-07-03:
This would indeed be interesting, but it would be great if you could share a bit more about what you are trying to do. In most programs, you'll need to interact with the environment to do anything useful.

Regarding `fd_write`: that's not JS specific, that's WASI. Is there a reason why having this import is a problem? You could probably just define this function in the embedder as a no-op.

One reason why TinyGo has many JS specific imports is because many packages define them when they target wasm. This is unfortunate, but unlikely to change until WASI is properly supported in Go.

dsherret, 2020-07-06:
> One reason why TinyGo has many JS specific imports is because many packages define them when they target wasm. This is unfortunate, but unlikely to change until WASI is properly supported in Go.

Oh, I see! So when I create a very basic file, say like so:

```go
package main

func main() {} // needed to compile

//go:export heavy_calulation
func heavy_calulation(a int, b int) int {
	return a + b // not heavy
}
```

I see the exports are:

```
[
  { name: "memory", kind: "memory" },
  { name: "memset", kind: "function" },
  { name: "_start", kind: "function" },
  { name: "__heap_base", kind: "global" },
  { name: "go_scheduler", kind: "function" },
  { name: "resume", kind: "function" },
  { name: "heavy_calulation", kind: "function" },
  { name: "./main.go.main", kind: "function" },
  { name: "__dso_handle", kind: "global" },
  { name: "__data_end", kind: "global" },
  { name: "__global_base", kind: "global" },
  { name: "__memory_base", kind: "global" },
  { name: "__table_base", kind: "global" }
]
```

And the imports are...

```
[ { module: "wasi_unstable", name: "fd_write", kind: "function" } ]
```

...so what I described in my original post does not actually apply since its one of my imports defining those functions. Is that correct? Yeah, that sucks packages do that—the end user compiling should be the that controls the public interface of the WASM module.

> This would indeed be interesting, but it would be great if you could share a bit more about what you are trying to do. In most programs, you'll need to interact with the environment to do anything useful.

The use case is calling a function exported from the wasm module (ex. `heavy_calculation` above) and using the returned value or reading from the module's memory. The module doesn't need to output anything. This is very easy to do in Rust by targeting `--wasm32-unknown-unknown` instead of `wasm32-wasi`.

In my case, I have a wasm interface described and plugins are implemented as wasm modules. If a wasm module implements the interface then it can be used as a plugin.

> Is there a reason why having this import is a problem? You could probably just define this function in the embedder as a no-op.

It's less of a problem if there's only one function, but IMO this import should only be required when targeting `wasm-wasi` as Rust does. It's noise to have to provide this when it isn't necessary and extra work to get Go compiled WASM files conforming to an interface that doesn't expect `fd_write`.

Basically, I think TinyGo should copy Rust:

* Have a `wasm32-unknown-unknown` target that has no imports and exports by default—it doesn't implement WASI and it doesn't require some Go specific interface. If someone wants to output they can implement their own import that is provided with a way to output.
* Have a `wasm32-wasi` target that compiles to a WASM file with WASI imports.

aykevl, 2020-07-07:
Okay, I guess we can do this. I don't think it will be very difficult, it mainly requires splitting up the runtime for the two different targets. See src/runtime/runtime_wasm.go. With build tags, you could split it into `wasi` and something else, not sure what to name it (`wasm32-unknown-unknown` is a LLVM target name, TinyGo names are usually just a board name).

deadprogram, 2021-02-21:
I think that the concerns in this issue have been most addressed in recent versions of TinyGo with WASI support. Can we close this particular issue now in favor of more specific ones?

deadprogram, 2021-03-28:
Now closing as mentioned in previous comment. Please reopen if needed. Thanks.

closed 2021-03-28 by deadprogram