tinygo-org/tinygo #1190: Allow targets to define more than 128 pins

firelizzard18, 2020-06-25:
Per discussion on https://github.com/tinygo-org/tinygo/pull/863#discussion_r441586516, `machine.Pin` needs to change to facilitate targets that have more than 128 pins.

> That may have an effect on many other targets. I think it's best to switch the `machine.Pin` type to `uint8` (and `machine.NoPin` to 0xff) which should also fix issues for a few other chips (like the stm32f7 family that often have a huge number of pins). This should be done in a separate PR, however.
>
> If `uint8` is not enough, the type should instead be chip family specific instead of changing to `int16` for all chips.

firelizzard18, 2020-06-25:
@aykevl I prefer chip family specific `type Pin`. That provides the most flexibility. Do you have a specific idea on how this should be done? One way would be requiring every target to define it, as in `type PinMode`. Another would be creating `machine_pin.go` and allowing targets to opt-out of the default by adding `// +build !my_target` to that file.

aykevl, 2020-06-30:
I wrote a PR to address this issue: https://github.com/tinygo-org/tinygo/pull/1201
It changes the Pin type for all boards at once, which (to my surprise) may lead to a small reduction in binary size.

> I prefer chip family specific `type Pin`. That provides the most flexibility.

What kind of flexibility? Using the same type everywhere also has benefits (no surprising behaviors across different chip families).

> One way would be requiring every target to define it, as in `type PinMode`.

Yes, if the Pin type is defined per family that would be the way to go in my opinion.

Sidenote: it may be a good idea to centralize that type as well. When I originally defined it, I expected more variation across GPIO pins but they are all very similar.

firelizzard18, 2020-07-01:
> What kind of flexibility? Using the same type everywhere also has benefits (no surprising behaviors across different chip families).

I generally dislike constraining runtime/machine implementations for wildly different chip families into the same set of concrete types. I much prefer interface-driven design, such as requiring `machine.UART` to conform to an interface, instead of requiring it to have specific fields and pass-by-value semantics.

aykevl, 2020-07-04:
> I generally dislike constraining runtime/machine implementations for wildly different chip families into the same set of concrete types. I much prefer interface-driven design, such as requiring [...]

Fair enough. But it really depends on the case. UARTs vary quite a bit in functionality across chips even though the basics (sending/receiving data) is more or less the same. In contrast, GPIO is much more limited, and the `Pin` type is intended as a universal way to identify a particular pin on a chip. Yes, if needed it can vary by chip but honestly I don't see the reason yet, if a switch to `uint8` fixes the 127 pins problem. (And note that this can always be changed again in the future to make it chip family specific, if there is a practical reason for it).

closed 2020-07-08 by firelizzard18