tinygo-org/tinygo #1192: Wasm build error

lu4p, 2020-06-25:
I tried installing tinygo from the AUR and the official arch repositories. On both I get this error when trying to build for wasm.
```
$ tinygo build -o wasm.wasm -target wasm ./main.go                     
error: failed to link /tmp/tinygo303515181/main: none of these commands were found in your $PATH: wasm-ld-10 wasm-ld
```

aykevl, 2020-06-30:
> I tried installing tinygo from the AUR and the official arch repositories.

Are you sure? The package in AUR (this one: https://aur.archlinux.org/packages/tinygo-bin/) should not have a `wasm-ld` dependency.
The non-AUR package most likely does require `wasm-ld` as the package size is too small to contain everything. Therefore it must be installed separately. I suspect the package does not fully specify the dependencies, but we can't solve that. Perhaps you can raise an issue with the maintainer?

In the meantime, you can probably solve this by installing the `lld` package manually.

@johanbrandhorst FYI

johanbrandhorst, 2020-06-30:
I recommend raising an issue against https://www.archlinux.org/packages/community/x86_64/tinygo/. Also TIL we've got an official arch package :tada:.

lu4p, 2020-06-30:
Filed a bug report: https://bugs.archlinux.org/task/67154

aykevl, 2020-06-30:
Thank you!
Closing this issue as this is a package bug, not a TinyGo bug.

closed 2020-06-30 by aykevl

lu4p, 2020-07-02:
￼ Comment by Anatol Pomozov (anatolik) - Thursday, 02 July 2020, 19:23 GMT+2 

added lld dependency in r657501. The next package build will contain the fix.