tinygo-org/tinygo #1194: tinygo make error undefined reference to Z3_mk_config

cornelk, 2020-06-26:
on a new system after building llvm using `make llvm-build`, I get the following error when running `make`:

```
# github.com/tinygo-org/tinygo/builder
/usr/bin/ld: tinygo/llvm-build/lib/libLLVMSupport.a(Z3Solver.cpp.o): in function `llvm::CreateZ3Solver()':
Z3Solver.cpp:(.text._ZN4llvm14CreateZ3SolverEv+0x2e): undefined reference to `Z3_mk_config'
/usr/bin/ld: Z3Solver.cpp:(.text._ZN4llvm14CreateZ3SolverEv+0x47): undefined reference to `Z3_set_param_value'
/usr/bin/ld: Z3Solver.cpp:(.text._ZN4llvm14CreateZ3SolverEv+0x5d): undefined reference to `Z3_set_param_value'
/usr/bin/ld: Z3Solver.cpp:(.text._ZN4llvm14CreateZ3SolverEv+0x73): undefined reference to `Z3_set_param_value'
/usr/bin/ld: Z3Solver.cpp:(.text._ZN4llvm14CreateZ3SolverEv+0x7b): undefined reference to `Z3_mk_context_rc'
/usr/bin/ld: Z3Solver.cpp:(.text._ZN4llvm14CreateZ3SolverEv+0x8a): undefined reference to `Z3_del_config'
/usr/bin/ld: Z3Solver.cpp:(.text._ZN4llvm14CreateZ3SolverEv+0x99): undefined reference to `Z3_set_error_handler'
/usr/bin/ld: Z3Solver.cpp:(.text._ZN4llvm14CreateZ3SolverEv+0xa1): undefined reference to `Z3_mk_simple_solver'
/usr/bin/ld: Z3Solver.cpp:(.text._ZN4llvm14CreateZ3SolverEv+0xf6): undefined reference to `Z3_solver_inc_ref'
```

I was able to fix it by adding `LLVM_OPTION += '-DLLVM_ENABLE_Z3_SOLVER=OFF'` to the makefile. Let me know if this is correct and I should submit a PR.

cornelk, 2020-06-26:
#889 could be solved by adding this as well. 

PR https://github.com/tinygo-org/tinygo/pull/1178/ contains a fix for this, merging it would solve it as well.

deadprogram, 2020-06-30:
#1178 has now been merged into `dev`, so marking for next release.

cornelk, 2020-06-30:
works now

closed 2020-06-30 by cornelk