tinygo-org/tinygo #1199: compiler: remove `-opt 0`

jaddr2line, 2020-06-28:
It doesn't actually work right now, so we might as well get rid of or disable it until it does.

aykevl, 2020-07-03:
Or fix it. I think it could be useful for debugging (even if the firmware is probably too big to run on a MCU).

But yes, I agree. We shouldn't "support" anything that doesn't actually work.

kenbell, 2021-03-14:
FYI - current status:
```
% tinygo help
...
  -opt string
        optimization level: 0, 1, 2, s, z (default "z")
...

% tinygo build --target bluepill -o bugtest.elf --opt 0 ./src/examples/bugtest
error: unknown optimization level: -opt=0
```

aykevl, 2021-03-14:
I have a fix here: https://github.com/tinygo-org/tinygo/pull/1689

deadprogram, 2021-05-12:
This was released with v0.18 so now closing. Thank you!

closed 2021-05-12 by deadprogram