tinygo-org/tinygo #1200: Backend for bytecode machine?

assyrianic, 2020-06-29:
I've developed my own virtual machine and I would like to create a backend for TinyGo that can target it.
The VM is a 64-bit RISC stack & register based minimal runtime environment.

The architecture is similar to RISC and an SSA type of IR where registers are based on an operand stack and allocated as necessary, able to use up to 256 registers at one time, here's an example implementing factorial:

```go
func factorial(i int) int {
    if i<=1 {
        return 1
    }
    return i * factorial(i-1)
}
```

```asm
factorial {
    pushlr
    alloc   3
    mov     r0, r3
    
;; if( i<=1 )
    movi    r1, 1
    ule     r0, r1
    jz      .L1
    
;; return 1;
    mov     r0, r1
    jmp     .L2
    
.L1
;; return i * factorial(i-1);
    mov     r2, r0    ;; int temp = i;
    sub     r2, r1    ;; temp -= 1;
    mov     r0, r2
    call    factorial ;; int res = factorial(temp);
    mul     r3, r0    ;; i * res;
    
.L2
    poplr
    redux   3
    
    ret
}
```

jaddr2line, 2020-06-30:
This should be doable once you create an LLVM backend for it. https://llvm.org/docs/WritingAnLLVMBackend.html

assyrianic, 2020-06-30:
that actually gave me jaddr, I could do either an LLVM IR to Tagha ASM or even the reverse as a sort of "JIT"?

aykevl, 2020-06-30:
Yes, writing a custom LLVM backend would be the way to go. In essence they convert LLVM IR to machine code, but LLVM backends also do some optimizations (including some target independent optimizations). The WebAssembly backend may be a good starting point, as it is also a (mostly) stack based machine.

Is this just to experiment or is this intended as something bigger?

assyrianic, 2020-06-30:
@aykevl it's part of my project to produce the fastest possible bytecode interpreter without using a JIT compiler. Thing is, what good is a bytecode interpreter runtime if there are no compilers or even backends that target it?

Also, I'm not sure if the WebAssembly backend would be a good starting point since my VM directly uses the stack as a dynamic register file. I'm going to attempt writing an LLVM IR => Tagha ASM transpiler and perhaps create a backend later.

closed 2024-12-02 by deadprogram