tinygo-org/tinygo #1205: wasm compiler cannot export a function that returns i64

moznion, 2020-06-30:
Hello,

It seems wasm compiler cannot export a function that returns i64. Reproduction code is the following:

```go
package main

//export returnsInt64
func returnsInt64() int64

func main() {
	returnsInt64()
}
```

then, compile this by `tinygo build -o main.wasm -target wasm ./main.go`, it generates like the following wasm:

```wat
(module
  (type (;0;) (func (param i32)))
  (type (;1;) (func (param i32 i32 i32 i32) (result i32)))
  (type (;2;) (func))
  (type (;3;) (func (param i32 i32 i32) (result i32)))
  (type (;4;) (func (param i32 i32)))
  (import "env" "returnsInt64" (func $returnsInt64 (type 0)))
  (import "wasi_unstable" "fd_write" (func $fd_write (type 1)))
  (func $__wasm_call_ctors (type 2))
...
```

This shows `returnsInt64` is a function that receives `i32` and no value returns. Is this an expected behavior?

And my tinygo environment is here:

```
$ tinygo version
tinygo version 0.13.1 darwin/amd64 (using go version go1.14.2 and LLVM version 10.0.1)
```

moznion, 2020-07-01:
I tried compiling with `-wasm-abi generic`, it builds the wasm with a correct signature. Should I use this option?

aykevl, 2020-07-03:
That depends on what you want to use it for. By default, TinyGo transforms `i64` parameters so that the value is passed by reference instead of by value (hence the i32 which is a pointer), because JavaScript previously did not support the `i64` type directly. It appears that this is now support in Chrome at least so once all popular browsers support it, we can switch to native `i64` types.

If you use TinyGo WebAssembly output in a non-browser environment, then `-wasm-abi=generic` is probably what you are looking for.

moznion, 2020-07-07:
I understand that. Thank you for the information.

moznion, 2020-07-07:
However, it transpiled `returnsInt64()` to `func (param i32)` wasm (not `func (result i32)`) in this case. I feel that was a little strange... anyway, this should be another issue.

closed 2020-07-07 by moznion

aykevl, 2020-07-07:
That `i32` parameter is a pointer where the return value will be written. This may seem strange, but this pattern is actually quite common in calling conventions. The reason is that it's usually much easier for the caller to reserve some area on the stack and pass a pointer than it is for the callee to reserve memory: where should this memory be allocated? It can't be allocated on the stack because the stack frame is gone after returning.

moznion, 2020-07-08:
Ah, it makes sense! Thank you for letting me know that.