tinygo-org/tinygo #1206: Process hangs when calling slice of functions in wasm

asdine, 2020-07-01:
Given this code:

```go
type foo struct {
	funcs []func()
}

func (f foo) do() error {
	for _, fn := range f.funcs {
		fn()
	}

	return nil
}

...
// this code is initialized somewhere 
var f foo

// this code is called from JS
err := f.do() // <- this hangs AFTER the do function has returned COMPLETELY
if err != nil {
	return err
}
```

If I remove the call to `fn` it doesn't hang anymore.

It's really hard for me to simplify my code and give you a working example, I wasn't able to reproduce this bug outside of my actual code.

I compiled using the 0.13.1 docker image with the following params:

docker run ... \
    tinygo/tinygo:0.13.1 tinygo build -o /dist/exp.wasm -target wasm --no-debug exp

aykevl, 2020-07-04:
> It's really hard for me to simplify my code and give you a working example, I wasn't able to reproduce this bug outside of my actual code.

Somebody will have to simplify it to fix the bug. Theoretically we could do it but you know the code better than we do.

What I usually do in such cases is simplify in small steps, in a copy of the code if needed because the changes will be big.
For example:

1. Eliminate external dependencies (imports). This is very important, as it massively reduces the area where the bug could be. If it seems to be in a dependency, try figuring out where in the dependency it is happening.
2. Simplify functions, perhaps starting by leaf functions (functions that don't call other functions). Try to inline functions when they become small, to make sure it isn't in the call somehow.
3. Remove every line that can be removed while not removing the bug.
4. Of the remaining code, try to simplify further. For example, replace an indirect call with a direct call. Use simpler control flow. Etc.

In this particular case, I suspect the problem may be in a blocking operation. You can create a trivially blocking operation using `time.Sleep`, which block but does not have side effects (apart from obviously waiting a bit).

deadprogram, 2021-04-22:
Closing due to inactivity. Please reopen if needed. Thank you!

closed 2021-04-22 by deadprogram