tinygo-org/tinygo #1207: does the reflect.Typeof and reflect.Valueof not working on wasm?

KamiD, 2020-07-02:
hello all, I wrote a json transfer code which is used to marshal or unmarshal from json string to interface{}
for test, I write a test code like:
```go
type InitMsg struct{
	Verifier string
	Beneficiary string
	Legacy 		int
}

func Test(){
   str := "Vals : " + reflect.ValueOf(initMsg).String() + " Type : " + reflect.TypeOf(initMsg).String()
   fmt.Println(str)
}
```

Output is 
```sh
"Vals : <T>  Type : T"
```

I want know that is reflect package have full support on wasm?

jaddr2line, 2020-07-02:
ValueOf and TypeOf are implemented, but not all methods on values and types are fully implemented yet. This is not wasm-specific.

aykevl, 2020-07-04:
> Output is
> 
> ```shell
> "Vals : <T>  Type : T"
> ```

Right now the `.String()` method simply returns `"T"`, to reduce the data size needed for reflection. That does not mean the type is not known. In fact, many parts of reflection already work, take a look at testdata/reflect.go to get an idea which part already work.
Is there a reason why you need to know the type name?

KamiD, 2020-07-06:
> Right now the `.String()` method simply returns `"T"`, to reduce the data size needed for reflection. That does not mean the type is not known. In fact, many parts of reflection already work, take a [...]

cool， it's very helpful

dgryski, 2023-03-23:
This work now.

closed 2023-06-14 by deadprogram