tinygo-org/tinygo #1209: Some digital pins not working correctly for atmega2560

brandentimm, 2020-07-04:
On Arduino Mega Rev 3 with ATmega2560, TinyGo has some incorrect digital pin mappings.

* Pin D2 is mapped to PE6, which should be PE4.  Reference for the chip is here: https://www.arduino.cc/en/Hacking/PinMapping2560
* Pin D5 is mapped to PE3, which is incorrectly calculated as (portE + 2) when it should be (portE + 3)

I have a working change against the dev branch for the two above issues.  Haven't yet submitted a PR because the contributor guide asks for an issue first.  Also, some unrelated wasm unit tests aren't passing for me, not sure if those are just currently broken or if I'm missing some dependencies for the test.  Here's the branch comparison for my changes: https://github.com/tinygo-org/tinygo/compare/master...brandentimm:atmega2560-fix-dpin-mapping?expand=1

Finally, as alluded to in #1154 digital pins 42-49 are not working.  Those are all on Port L.  

These lines are suspicious: https://github.com/tinygo-org/tinygo/blob/a9ba6ebad9ad85cacd9674a1af9229489a811f68/src/machine/machine_atmega2560.go#L90-L96

It seems PL7 should be (portL + 7) instead of (portE + 7).  However this change had no effect, and still doesn't explain why none of the PL* pins are working.

aykevl, 2020-07-05:
> I have a working change against the dev branch for the two above issues. Haven't yet submitted a PR because the contributor guide asks for an issue first.

For trivial changes / bugfixes that's not really necessary, it's really about large changes that we would like to have an issue for first to discuss whether the design is good or it's something we want in TinyGo. The main goal is to avoid unnecessary and duplicated work. So feel free to make a PR.

> It seems PL7 should be (portL + 7) instead of (portE + 7).

Yeah definitely, that looks like a copy-paste error. A PR with a fix would be welcome.
I don't know why it wouldn't work. Theoretically it could be related to bugs in the AVR backend, most of which should be fixed in LLVM 11 (coming in a few months). But my guess would be that it's something else.
You can try direct register accesses (and do the same in Arduino C) to get closer "to the metal" to debug. I don't have an atmega2560 so can't debug it myself.

brandentimm, 2020-07-05:
Thanks @aykevl , I've submitted PR #1211 with this change.

deadprogram, 2020-07-05:
#1211 has been merged, so marking this issues as "next-release". Thanks everyone!

brandentimm, 2020-07-06:
Thanks @deadprogram , I plan on working on the Port L issues later this week once the board is freed up from a current project.  I'll assume this can stay open, let me know if you want me to open a separate issue for that.

brandentimm, 2020-07-08:
PR to fix the Port L mapping issue referenced above #1222

deadprogram, 2020-08-07:
This was released with v0.14.0 so now closing. Please reopen if needed. Thanks!

closed 2020-08-07 by deadprogram