tinygo-org/tinygo #1210: SIGILL: illegal instruction on MacOS Catalina 10.15.5 on 2 GHz Intel Core 2 Duo

sglebs, 2020-07-04:
The install went ok, but it crashes:

```
tinygo
SIGILL: illegal instruction
PC=0x4064f40 m=0 sigcode=1

goroutine 1 [running, locked to thread]:
runtime.asyncPreempt()
	/usr/local/go/src/runtime/preempt_amd64.s:8 fp=0xc0000f7ef0 sp=0xc0000f7ee8 pc=0x4064f40
github.com/tinygo-org/tinygo/interp.init()
	/Users/distiller/project/interp/errors.go:16 fp=0xc0000f7ef8 sp=0xc0000f7ef0 pc=0x43663b0
runtime.doInit(0x8d9a400)
	/usr/local/go/src/runtime/proc.go:5414 +0x8a fp=0xc0000f7f28 sp=0xc0000f7ef8 pc=0x404307a
runtime.doInit(0x8d9c8a0)
	/usr/local/go/src/runtime/proc.go:5409 +0x57 fp=0xc0000f7f58 sp=0xc0000f7f28 pc=0x4043047
runtime.doInit(0x8d9c520)
	/usr/local/go/src/runtime/proc.go:5409 +0x57 fp=0xc0000f7f88 sp=0xc0000f7f58 pc=0x4043047
runtime.main()
	/usr/local/go/src/runtime/proc.go:190 +0x1ce fp=0xc0000f7fe0 sp=0xc0000f7f88 pc=0x40365de
runtime.goexit()
	/usr/local/go/src/runtime/asm_amd64.s:1373 +0x1 fp=0xc0000f7fe8 sp=0xc0000f7fe0 pc=0x40636e1

rax    0x8d9a400
rbx    0xa
rcx    0x0
rdx    0x8d9a450
rdi    0x0
rsi    0x43663b0
rbp    0xc0000f7f18
rsp    0xc0000f7ee8
r8     0x8dc5b01
r9     0xc0000427a8
r10    0xc0000427a0
r11    0xc0000427a4
r12    0x6
r13    0x5
r14    0x200
r15    0x0
rip    0x4064f40
rflags 0x10202
cs     0x2b
fs     0x0
gs     0x0
```

Mac mini (Early 2009)
2 GHz Intel Core 2 Duo
8 GB 1333 MHz DDR3
NVIDIA GeForce 9400 256 MB

sglebs, 2020-07-04:
Related? https://github.com/golang/go/issues/37459

jaddr2line, 2020-07-04:
What version is this?

jaddr2line, 2020-07-04:
Actually yeah, that Go issue would have to be it. The next release should use a newer version of Go with the fix. For now, you should be able to build Go from source using Go `<1.14` or `>=1.14.1`.

sglebs, 2020-07-04:
```
 go version
go version go1.14.4 darwin/amd64
```

sglebs, 2020-07-04:
Trying from sources.

```
git clone --recursive https://github.com/tinygo-org/tinygo.git
cd tinygo
go install
go: downloading go.bug.st/serial v1.0.0
go: downloading tinygo.org/x/go-llvm v0.0.0-20200401165421-8d120882fc7a
go: downloading github.com/blakesmith/ar v0.0.0-20150311145944-8bd4349a67f2
go: downloading github.com/marcinbor85/gohex v0.0.0-20180128172054-7a43cd876e46
go: downloading golang.org/x/tools v0.0.0-20200216192241-b320d3a0f5a2
go: downloading github.com/google/shlex v0.0.0-20181106134648-c34317bd91bf
go: downloading github.com/creack/goselect v0.1.1
go: downloading golang.org/x/sys v0.0.0-20191128015809-6d18c012aee9
# tinygo.org/x/go-llvm
../go/pkg/mod/tinygo.org/x/go-llvm@v0.0.0-20200401165421-8d120882fc7a/analysis.go:16:10: fatal error: 'llvm-c/Analysis.h' file not found
#include "llvm-c/Analysis.h" // If you are getting an error here read bindings/go/README.txt
         ^~~~~~~~~~~~~~~~~~~
1 error generated.
# github.com/tinygo-org/tinygo/cgo
cgo/libclang.go:18:10: fatal error: 'clang-c/Index.h' file not found
#include <clang-c/Index.h> // if this fails, install libclang-10-dev
         ^~~~~~~~~~~~~~~~~
1 error generated.
```

```
clang --version
Apple clang version 11.0.3 (clang-1103.0.32.62)
Target: x86_64-apple-darwin19.5.0
Thread model: posix
InstalledDir: /Library/Developer/CommandLineTools/usr/bin
```

Trying to figure out where this bindings/go/README.txt is for me to read it and try to get it to build from sources on my machine.

aykevl, 2020-07-05:
Yes, it looks like our previous release was built with the buggy Go version.

> Trying to figure out where this bindings/go/README.txt is for me to read it and try to get it to build from sources on my machine.

No, that's a misleading error message from upstream LLVM. TinyGo uses LLVM in a different way.

The easiest option is usually to install LLVM from Homebrew. It should usually work, but some people have reported issues with it. Please let us know how it goes.

```sh
brew install llvm@10
go install # in the tinygo directory
```

sglebs, 2020-07-05:
I am afraid of having a second LLVM. I guess I will have to wait for the next tinygo binary release :-(   In the meantime I will take a look at Gobot and MicroScheme :)  Thanks!!

deadprogram, 2020-07-06:
We should release a v0.13.2 that is built with a newer Go probably. What do you think @aykevl ?

aykevl, 2020-07-07:
> I am afraid of having a second LLVM. I guess I will have to wait for the next tinygo binary release :-(

I believe installing LLVM via Homebrew is pretty safe and should not interfere with Xcode. Alternatively, you can build it yourself, see https://tinygo.org/getting-started/macos/#with-a-self-built-llvm (but that will take a while, especially on this older CPU). Building LLVM yourself will not affect the existing installation.

> We should release a v0.13.2 that is built with a newer Go probably. What do you think @aykevl ?

I think we should just be doing a v0.14 soon, which will automatically fix this.

I've added the next-release tag, because whatever we do it will automatically be fixed with the next release.

sglebs, 2020-07-22:
What is the ETA for 0.14?

deadprogram, 2020-08-07:
This was released with v0.14.0 so now closing. Please reopen if needed. Thanks!

closed 2020-08-07 by deadprogram

sglebs, 2020-12-30:
Still fails for me , tinygo 0.16:

```
marcio$ tinygo --version
SIGILL: illegal instruction
PC=0x40654f0 m=0 sigcode=1

goroutine 1 [running, locked to thread]:
runtime.asyncPreempt()
	/usr/local/go/src/runtime/preempt_amd64.s:8 fp=0xc000043ec0 sp=0xc000043eb8 pc=0x40654f0
math/big.init()
	/usr/local/go/src/math/big/arith_amd64.go:11 fp=0xc000043ec8 sp=0xc000043ec0 pc=0x41306a0
runtime.doInit(0x8debb20)
	/usr/local/go/src/runtime/proc.go:5414 +0x8a fp=0xc000043ef8 sp=0xc000043ec8 pc=0x40435da
runtime.doInit(0x8deb1e0)
	/usr/local/go/src/runtime/proc.go:5409 +0x57 fp=0xc000043f28 sp=0xc000043ef8 pc=0x40435a7
runtime.doInit(0x8dec360)
	/usr/local/go/src/runtime/proc.go:5409 +0x57 fp=0xc000043f58 sp=0xc000043f28 pc=0x40435a7
runtime.doInit(0x8ded0a0)
	/usr/local/go/src/runtime/proc.go:5409 +0x57 fp=0xc000043f88 sp=0xc000043f58 pc=0x40435a7
runtime.main()
	/usr/local/go/src/runtime/proc.go:190 +0x1ce fp=0xc000043fe0 sp=0xc000043f88 pc=0x4036b3e
runtime.goexit()
	/usr/local/go/src/runtime/asm_amd64.s:1373 +0x1 fp=0xc000043fe8 sp=0xc000043fe0 pc=0x4063c91

rax    0x8debb20
rbx    0xd
rcx    0x0
rdx    0x8debb88
rdi    0x30
rsi    0x41306a0
rbp    0xc000043ee8
rsp    0xc000043eb8
r8     0xe80320e
r9     0x203000
r10    0x4
r11    0x32
r12    0xf2
r13    0x0
r14    0x7e00be6
r15    0x0
rip    0x40654f0
rflags 0x10202
cs     0x2b
fs     0x0
gs     0x0

marcio$ go version
go version go1.14.4 darwin/amd64
MacMini-Vintage:microscheme marcio$ 
```

sglebs, 2020-12-30:
I am unable to reopen this issue. But the problem persists.

sglebs, 2020-12-30:
It looks like it needs to be built with Go 1.14.10: https://github.com/pgaskin/kepubify/commit/b860308a724219a6e33ee4f952fbadc25fd98cad