tinygo-org/tinygo #1212: `gen-device-nrf` is failing

ericonr, 2020-07-06:
I pulled from latest master:

```
$ make gen-device-nrf
./build/gen-device-svd -source=https://github.com/NordicSemiconductor/nrfx/tree/master/mdk lib/nrfx/mdk/ src/device/nrf/
cannot find input directory: lib/nrfx/mdk/
make: *** [Makefile:133: gen-device-nrf] Error 1
```

Something must have changed in Nordic's repo, which broke generation here. So, I think this is both a bug and a feature request: would it be possible to have the `gen-device` targets have some way of "locking" the version they are going to use, so that both packaging is reproducible and arbitrary failures can't happen?

deadprogram, 2020-07-06:
You probably need to run `git submodule update --init` first.

ericonr, 2020-07-07:
Oof, yeah, indeed. Thanks for the quick response! I will see if I can make a PR to make that requirement clearer, would you be open to that?

deadprogram, 2020-07-07:
See https://github.com/tinygo-org/tinygo/pull/1213

aykevl, 2020-07-07:
Also see #1219. It includes `git clone --recursive`, which will also download submodules.

ericonr, 2020-07-08:
I think this has been reasonably solved, thanks!

closed 2020-07-08 by ericonr