tinygo-org/tinygo #1214: runtime: allow interrupts to wake the scheduler

jaddr2line, 2020-07-06:
In order for interrupt-based wake-ups (as introduced by #1142) to work concurrently with `time.Sleep`, we need to make some per-board changes.

Previously, `sleepTicks` (the function used as an interface between the scheduler and the hardware timer) was defined to block until the timer completed, since there was nothing else to do. Now we need to change this so that it bails out when an interrupt wakes the chip. I started with the 2 main chip types I have (`atsamd21` and `atsamd51`). This change is trivial, but requires testing on all supported platforms.

My `atsamd51` fix can be used as a reference for how this works: https://github.com/tinygo-org/tinygo/pull/1197/files

If you want to try to port over the fix to a board class, comment on this thread to claim it so we don't duplicate work.

Platforms:

- [x] wasm
- [x] arm7tdmi (GBA) (no actual timer implemented yet so I guess this is done)
- [ ] avr
- [ ] stm32f103xx
- [ ] stm32f407
- [ ] nrf
- [ ] fe310
- [x] atsamd21
- [x] atsamd51
- [ ] mk66f18 (new board in #863)

firelizzard18, 2020-07-08:
I updated #863 with support for this

kenbell, 2022-06-26:
This is an oldie - but I know for sure this is done for RP2040 and STM32