tinygo-org/tinygo #1224: [BUG] Compile error when value type of a map is byte or int8

latiif, 2020-07-09:
I originally came across this bug when compiling this snippet:

```go
package  main
import  "fmt"
var myMap =  map[rune]byte{'n': '\n'}
func  main() {
	fmt.Println("Hello World")
}
```
The line in question is 
```go
var myMap =  map[rune]byte{'n': '\n'}
```
which ,as you can see, declares a map from `rune` to `byte`. 
I got the following output: [log.txt](https://github.com/tinygo-org/tinygo/files/4897620/log.txt)

After further investigation, the issue seems to persist whenever the value type of the map is `byte` (`uint8`) or `int8`.
```go
var myMap =  map[string]byte{"mystring": '\n'} // fails
var myMap = map[interface{}]byte{"mystring": '\n'} //fails
var myMap = map[int]byte{13: 'b'} // fails
```
When the map is empty it runs ok.
```go
var myMap = map[interface{}]byte{} // compiles
```
When the map key type is `byte` it also seems to be ok
```go
var myMap = map[byte]string{'\n': "mystring"} // compiles
var myMap = map[byte]interface{}{'\n': "mystring"} // compiles
var myMap = map[byte]int{'b': 13} // compiles
```

While googling I came across [this issue](https://github.com/hybridgroup/gdg-2019/issues/6) that mentions `tinygo` and #787.

`tinygo` version: `tinygo version 0.13.1 linux/amd64 (using go version go1.14.3 and LLVM version 10.0.1)`

deadprogram, 2020-09-12:
This is still an issue with the latest `dev` branch:

```
$ tinygo version
tinygo version 0.15.0-dev-9448995 linux/amd64 (using go version go1.15 and LLVM version 10.0.1)

$ tinygo run ./src/examples/try/try.go
fatal error: unexpected signal during runtime execution
[signal SIGSEGV: segmentation violation code=0x1 addr=0x28 pc=0x48d0730]

runtime stack:
runtime.throw(0x4c40850, 0x2a)
        /home/ron/.gvm/gos/go1.15/src/runtime/panic.go:1116 +0x72
runtime.sigpanic()
        /home/ron/.gvm/gos/go1.15/src/runtime/signal_unix.go:704 +0x4ac

goroutine 1 [syscall]:
runtime.cgocall(0xd56590, 0xc0048065c0, 0xc00003b000)
        /home/ron/.gvm/gos/go1.15/src/runtime/cgocall.go:133 +0x5b fp=0xc004806590 sp=0xc004806558 pc=0xa1733b
tinygo.org/x/go-llvm._Cfunc_LLVMGetConstOpcode(0x7fc43c8233f8, 0x7fc400000000)
        _cgo_gotypes.go:5514 +0x49 fp=0xc0048065c0 sp=0xc004806590 pc=0xc57a09
tinygo.org/x/go-llvm.Value.Opcode.func1(0x7fc43c8233f8, 0xc004806620)
        /home/ron/.gvm/pkgsets/go1.15/global/pkg/mod/tinygo.org/x/go-llvm@v0.0.0-20200503225853-345b2947b59d/ir.go:899 +0x4d fp=0xc0048065f0 sp=0xc0048065c0 pc=0xc71c6d
tinygo.org/x/go-llvm.Value.Opcode(0x7fc43c8233f8, 0xc7118d)
        /home/ron/.gvm/pkgsets/go1.15/global/pkg/mod/tinygo.org/x/go-llvm@v0.0.0-20200503225853-345b2947b59d/ir.go:899 +0x2b fp=0xc004806610 sp=0xc0048065f0 pc=0xc638cb
github.com/tinygo-org/tinygo/interp.(*MapValue).PutBinary(0xc004a9ef50, 0xc0059b9890, 0xc0059b9860, 0xc004807230, 0x1)
        /home/ron/Development/tinygo/tinygo/interp/values.go:372 +0x5b fp=0xc0048066a8 sp=0xc004806610 pc=0xcf5cfb
github.com/tinygo-org/tinygo/interp.(*frame).evalBasicBlock(0xc004807160, 0x7fc4407b6810, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, ...)
        /home/ron/Development/tinygo/tinygo/interp/frame.go:342 +0x4f85 fp=0xc0048070d0 sp=0xc0048066a8 pc=0xce9085
github.com/tinygo-org/tinygo/interp.(*evalPackage).function(0xc004807368, 0x7fc4380360e8, 0xc004807380, 0x2, 0x2, 0x0, 0x0, 0x0, 0x0, 0x0)
        /home/ron/Development/tinygo/tinygo/interp/interp.go:113 +0x249 fp=0xc004807288 sp=0xc0048070d0 pc=0xcf1749
github.com/tinygo-org/tinygo/interp.Run(0x67aa890, 0x0, 0x0, 0x0)
        /home/ron/Development/tinygo/tinygo/interp/interp.go:85 +0x7e7 fp=0xc0048074f0 sp=0xc004807288 pc=0xcf1187
github.com/tinygo-org/tinygo/builder.Build(0x7ffc528d5fbe, 0x19, 0x4c2d8b0, 0x4, 0xc0001c1e00, 0xc004807bb0, 0x0, 0x0)
        /home/ron/Development/tinygo/tinygo/builder/build.go:65 +0x21a fp=0xc004807b70 sp=0xc0048074f0 pc=0xd2129a
main.Run(0x7ffc528d5fbe, 0x19, 0xc000286000, 0xe, 0x0)
        /home/ron/Development/tinygo/tinygo/main.go:434 +0xb1 fp=0xc004807bd0 sp=0xc004807b70 pc=0xd33fb1
main.main()
        /home/ron/Development/tinygo/tinygo/main.go:912 +0x10a9 fp=0xc004807f88 sp=0xc004807bd0 pc=0xd370c9
runtime.main()
        /home/ron/.gvm/gos/go1.15/src/runtime/proc.go:204 +0x209 fp=0xc004807fe0 sp=0xc004807f88 pc=0xa4b189
runtime.goexit()
        /home/ron/.gvm/gos/go1.15/src/runtime/asm_amd64.s:1374 +0x1 fp=0xc004807fe8 sp=0xc004807fe0 pc=0xa7ca61
ron@ron-XPS-13-7390:~/Development/tinygo/tinygo (dev)
```

kenbell, 2021-03-14:
FYI - this seems to be fixed.  This works on latest dev, with the code above
```
go install                                                          
tinygo version
tinygo version 0.18.0-dev darwin/amd64 (using go version go1.15.8 and LLVM version 11.0.0)
tinygo clean
tinygo build --target bluepill -o bugtest.elf ./src/examples/bugtest
```

aykevl, 2021-03-14:
This was probably fixed as part of #1430.

deadprogram, 2021-03-14:
OK closing. Please reopen if needed.

closed 2021-03-14 by deadprogram