tinygo-org/tinygo #1228: d.WriteByte() undefined for Arduino Nano and Bluepill

alankrantas, 2020-07-10:
I'm testing NeoPixel (WS2812) modules on some boards, and for Arduino Nano and Bluepill (for now I build the code and upload it via STLink Utility) the pin's WriteByte() are not defined:

```
go\src\tinygo.org\x\drivers\ws2812\ws2812.go:23:5: d.WriteByte undefined (type Device has no field or method WriteByte)
go\src\tinygo.org\x\drivers\ws2812\ws2812.go:32:5: d.WriteByte undefined (type Device has no field or method WriteByte)
go\src\tinygo.org\x\drivers\ws2812\ws2812.go:33:5: d.WriteByte undefined (type Device has no field or method WriteByte)
go\src\tinygo.org\x\drivers\ws2812\ws2812.go:34:5: d.WriteByte undefined (type Device has no field or method WriteByte)
```

Arduino Uno does has WriteByte() though.

deadprogram, 2020-07-11:
The assembly language required for this particular device is not available for every MCU. This is a somewhat unique characteristic of the WS2818 due to the highly precise timing it requires.

I do not think either the Arduino or STM32 are currently supported. We'd love to get a PR that adds it!

alankrantas, 2020-07-11:
I think WS2818 runs pretty well on Uno, and Nano is essentially a smaller Uno with an older bootloader. In fact - just tested it - I can build a hex file targeted for Uno and then use AVR to upload it to Nano.

[image]

Any tips for where should I look at? I haven't yet been able to figure it out.

deadprogram, 2020-07-11:
See https://github.com/tinygo-org/drivers/tree/master/ws2812 for TinyGo driver for WS2812.

aykevl, 2020-07-12:
In particular, look at https://github.com/tinygo-org/drivers/blob/master/ws2812/ws2812_avr_16m.go, where you can add the Nano to the list of build tags (now just `arduino`).

alankrantas, 2020-07-13:
Thanks, I made a PR at the Tinygo driver repository. Close this for now.

closed 2020-07-13 by alankrantas