tinygo-org/tinygo #1234: compiler: move goroutine start pointer pack to the stack

jaddr2line, 2020-07-11:


jaddr2line, 2020-07-11:
On `coroutines`, we really do not need to do anything special.

I currently see 3 reasonable techniques for this on `tasks`:
1. Copy the data to the bottom of the goroutine's stack. The goroutine start wrapper can then safely load it without caring about lifetime, since the data will have the same lifetime as the goroutine. However, this has the disadvantage that the section of stack can never be used for anything else.
2. Copy the data to the top of the goroutine's stack. This works as long as all of the data is loaded before that section is used.
3. Stack-allocate the data and synchronize the starting goroutine with the goroutine creating it. This would basically change `task.start` to block until the goroutine start wrapper had loaded all of the data (or if spawning from the system stack, this would call resume on the goroutine).

I think option 3 is probably the sanest version so far.

aykevl, 2020-07-15:
I don't think 1) is all that bad. Usually the number of parameters is very limited so the stack usage should be small. In fact, this is exactly what the main Go implementation does.
2) may be harder to do correctly. You need to make sure the data is placed at a point higher than `$gowrapper`, which might not be all that difficult (the data is there already if the stack size could be determined at compile time). Although, there is a chance that LLVM might pass some parameters to child functions leading to data corruption.
3. seems rather complex and may complicate the structure of the scheduler, which is unfortunate. It might make future optimizations harder to implement, for example.