tinygo-org/tinygo #1244: Question: Can tinygo perhaps be used to transpile Harfbuzz c++ to golang ?

joe-getcouragenow, 2020-07-14:
We need to port harfbuzz to golang for use in the golang GUI software called GIO. BTW we are currently are working towards cross compiling it for Web using Tinygo. We need this because the GUI is 100% just a Web Canvas.

This is the c++ code base. https://github.com/harfbuzz/harfbuzz

The Harfbuzz code base changes all the time.
I would prefer NOT to use CGO and Harfbuzz being in the Production code. 

So can we transpile it with Tinygo ?
Clang can read C++ and emit LLVM IR. TinyGo uses this to effect to be a golang compiler driven by LLVM IR.
MIght work ? 

Background about HarfBuzz:
BTW, google use HarfBuzz to support Font rendering in Skia where you have complex non latin languages.
E.G: https://github.com/aam/skiaex

jaddr2line, 2020-07-14:
> Clang can read C++ and emit LLVM IR. TinyGo uses this to effect to be a golang compiler driven by LLVM IR.
> Your pipeline seems to suggest it might: https://tinygo.org/compiler-internals/pipeline/
> Clang can produce LLMIR, that tinygo can then turn into binaries, but can it turn it into golang ?

I think you may be somewhat misunderstanding some of this. TinyGo does not does not turn LLVM IR into anything. It converts Go code into LLVM IR, transforms it a bit, then passes it off to some LLVM tools to be optimized and translated into machine code. The only place clang is involved is CGO. TinyGo uses clang to parse C function declarations from headers which it uses to generate CGO bindings, and then also uses clang to build C(++) files in a Go package.

If you wanted to turn LLVM IR into Go, that would be extremely difficult. LLVM IR basically looks like C, except with a lot of things behaving slightly different, and adding some C++ stuff (e.g. C++ exception handling). There was a C++ LLVM backend for a while, but it eventually had to be removed from LLVM because it no longer worked and consistently required tons of maintenance because this conversion was non-trivial. Converting LLVM IR to Go would probably not be practical, as this means re-implementing C++ exceptions, manual memory management, etc. This is not really a goal of TinyGo. TinyGo aims to have good interoperability with C, not to replace it.

> The Harfbuzz code base changes all the time.
> I would prefer NOT to use CGO and Harfbuzz being in the Production code.

CGO is probably your best choice. You can pin a specific version of Harfbuzz in a build script if you want. In TinyGo, CGO behaves like a regular C function call, because it really is one. TinyGo forward declares a function in C calling convention in the IR and inserts a call to it.

> Which begs the question: " Can i use Tinygo to compile golang code for x86 etc etc". Just wondering as this would mean no GC and all the follow on goody effects...

TinyGo has a GC (actually 2, soon to be 3 different GCs which you can choose from).

joe-getcouragenow, 2020-07-14:
Thankyou for the in depth explanation.

So if I use golang calling via cgo to c++ will tinygo be able to compile
all into wasm to produce a single WASM for web ?

Likewise, on embedded, will tinygo do the same thing and convert it all to
native binary ?

jaddr2line, 2020-07-14:
TinyGo does not convert anything. You need to build libharfbuzz somehow for whatever target you want to support, and then TinyGo can link against it with CGO. However, libharfbuzz appears to be fairly large (on my system it looks like it is taking up 941 KiB), so it will most likely not fit on many embedded systems.

joe-getcouragenow, 2020-07-14:
> TinyGo does not convert anything. You need to build libharfbuzz somehow for whatever target you want to support, and then TinyGo can link against it with CGO. However, libharfbuzz appears to be [...]

For Web this is not going to work then.  The onyl way to do it seems to be to compile the c++  libharfbuzz to wasm, and then the main golang lib ( mainlib ).
But then your back to how to get WASM mainlib to call WASM libharfbuzz..  :)

aykevl, 2020-07-15:
I'm not sure what your goal is here. If you want to transpile C to Go, then there may be tools for that (TinyGo is not one of them, you could take a look at [c2go](https://github.com/elliotchance/c2go)). If you just want to use it as a library, then you shouldn't transpile it but instead use CGo. Even then, I don't see an added benefit of using TinyGo for this library unless you actually want to write some of the program in Go.

> The only way to do it seems to be to compile the c++ libharfbuzz to wasm, and then the main golang lib ( mainlib ).

What is this mainlib? Do you mean the Go standard library?

joe-getcouragenow, 2020-07-18:
Thanks for not giving up.

The goal is to get hafbuzz usable for pure golang programs with no cgo.
This allows golang frameworks to have best in class font and
internationalisation features.

There is a discussion about this going on at the gio list and I was just
trying to brain storm using tinygo to help as I knew that c2go also uses
clang and LLVM aspects.

Check it out : https://todo.sr.ht/~eliasnaur/gio/146#comment-12818

deadprogram, 2023-05-23:
TinyFont can display non-latin languages like Japanese. Not sure if that was true when this issues was originally opened? In any case, now closing since it seems to have been addressed and is also inactive. Please open if needed.

closed 2023-05-23 by deadprogram