tinygo-org/tinygo #1246: can only convert from a string to a slice

louisliu2048, 2020-07-15:
when using tinygo to build project, errors return:

```
tinygo build -o main.wasm -target wasm ./main.go
panic: can only convert from a string to a slice

goroutine 1 [running]:
github.com/tinygo-org/tinygo/compiler.(*builder).createConvert(0xc003c7f5e8, 0x7dd77e0, 0xc001f71950, 0x7dd78a0, 0xc000c236d0, 0x3603bb40, 0x103090, 0x4005b23, 0x9f25bf0, 0x0)
	/Users/distiller/project/compiler/compiler.go:2571 +0x1b7e
github.com/tinygo-org/tinygo/compiler.(*builder).createExpr(0xc003c7f5e8, 0x7de64c0, 0xc004f200f0, 0x7de64c0, 0xc004f200f0, 0x1301)
	/Users/distiller/project/compiler/compiler.go:1543 +0x1441
github.com/tinygo-org/tinygo/compiler.(*builder).createInstruction(0xc003c7f5e8, 0x7de6460, 0xc004f200f0)
	/Users/distiller/project/compiler/compiler.go:1055 +0x1474
github.com/tinygo-org/tinygo/compiler.(*builder).createFunctionDefinition(0xc003c7f5e8)
	/Users/distiller/project/compiler/compiler.go:1015 +0xed4
github.com/tinygo-org/tinygo/compiler.Compile(0x7ffeefbff58f, 0x9, 0x9f05620, 0xc00021e240, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, ...)
	/Users/distiller/project/compiler/compiler.go:302 +0x11a4
github.com/tinygo-org/tinygo/builder.Build(0x7ffeefbff58f, 0x9, 0x7ffeefbff578, 0x9, 0xc00021e240, 0xc003c7fcc0, 0x0, 0x0)
	/Users/distiller/project/builder/build.go:35 +0x9e
main.Build(0x7ffeefbff58f, 0x9, 0x7ffeefbff578, 0x9, 0xc0001900e0, 0x0, 0x12)
	/Users/distiller/project/main.go:97 +0xc5
main.main()
	/Users/distiller/project/main.go:806 +0x192f
make: *** [build] Error 2
```

the error info tells me that it failed to *createFunctionDefinition* , no more info. I have no ideal what's wrong with my code!

louisliu2048, 2020-07-15:
according to panic info, check my code, find out that I defined an alia type for string, and tinygo cannot figure it out.

aykevl, 2020-07-16:
Can you please give the code that causes this error? That allows us to reproduce and fix the bug.

louisliu2048, 2020-07-27:
> Can you please give the code that causes this error? That allows us to reproduce and fix the bug.

sorry for reply so late, here is the code, although it is not the source code, the error can be reproduced:

```
package main

import (
	"fmt"
)

type HumanAddress string

func CanonicalAddress(human HumanAddress) ([]byte, error) {

	// ret := C.canonicalize_address(unsafe.Pointer(regionHuman), unsafe.Pointer(regionCanon))
	// C.free(humanPtr)
	//
	// if ret < 0 {
	// 	return nil, errors.New("canonicalize_address returned error")
	// }
	// canoAddress := TranslateToSlice(uintptr(regionCanon))

	canoAddress := []byte(human)

	return canoAddress, nil

}

func main() {
	hAddr := "test address"
	cAddr, err := CanonicalAddress(HumanAddress(hAddr))
	if err != nil {
		fmt.Println(err)
		return
	}

	fmt.Println(string(cAddr))
}
```

aykevl, 2020-07-27:
Thank you! That's even better (a small standalone reproducer). Here is a fix: https://github.com/tinygo-org/tinygo/pull/1269

jbpratt, 2020-11-23:
This can be closed! The repro compiles successfully.

deadprogram, 2020-11-24:
Awesome, thanks for verifying @jbpratt

closed 2020-11-24 by deadprogram