tinygo-org/tinygo #1250: proposal: add cli command to get device info

nrwiersma, 2020-07-16:
Thanks for a great project.

It would be a useful addition to allow the CLI to get device information, including a unique id for the device, given a target and port. 

It is not uncommon to have multiple devices that need unique information per device. Go tags solves the issue for the most part, just by adding a build tag to the files, but it is still required to set the correct tag for the device. This can be error prone. With something like `tinygo device-info -target XXX -port YYY` and some Makefile magic, the device id could be used as the Go build tag.

aykevl, 2020-07-16:
This is extremely device specific, and many chips don't even have a unique ID. When they have one, I'm not sure it's that easy to get it from the device without using a debugger (or uploading a custom piece of code specifically to get the device ID). My solution so far is to write a `Makefile` with multiple targets and using the right one for the right device.

While this is a good idea, I'm not sure how feasible this is and how commonly this is used. I think it's feasible for many devices, but I'm not sure whether the cost is worth the benefit (unless someone really wants to have it and wants to put in the effort).

nrwiersma, 2020-07-16:
For the most part I agree. I spent several hours trying to get it for an Arduino Nano. It does exists, but seems to be a hidden feature that is not well supported. I am already using a Makefile with vars to achieve this, but seemed like a nice feature to alleviate some pain and error.

I also dont know how often this would be used. Perhaps this is better as a stand alone tool for specific hardware.

Appreciate you taking the time to have a look.

closed 2020-07-16 by nrwiersma