tinygo-org/tinygo #1255: avr: timer stops sleep from working

nrwiersma, 2020-07-20:
I am using an arduino nano, so avr with no scheduler. I am playing to keeping time using timer0. When timer0 is enabled, the sleep is woken by the timer. My program looks like:

```go
package main

import (
	"strconv"
	"time"

	"github.com/nrwiersma/ard-timex"
)

func main() {
	for {
		now := timex.Now()
		println("Time: ", strconv.FormatInt(now, 10))

		time.Sleep(time.Second)
	}
}
```

`timex` is running timer0 with the following config:
```go
avr.TCCR0B.Set(avr.TCCR0B_CS00 | avr.TCCR0B_CS01)
avr.TIMSK0.Set(avr.TIMSK0_TOIE0)
avr.TIFR0.Set(avr.TIFR0_TOV0)
```

The program prints `Time: xxx` way more often than once a second, closer to 4 times a second.

jaddr2line, 2020-07-20:
Related to #952 and #1154

nrwiersma, 2020-07-20:
I briefly looked at implementing timer0 for keeping time on AVR, but the boards used a mixed bag of `TCCR0B`, `TCCR0` and `TCCR0A` with no real delineation lines that could be used for build tags, and Go lacks the `#ifdef` that is used in C++ to make this work decently in the Arduino framework. 

Also I seem to be incapable of making the dev branch work on my mac, but not for a lack of trying.

aykevl, 2020-07-20:
> I briefly looked at implementing timer0 for keeping time on AVR,

That would certainly work, but it would interfere with PWM on two pins.

> but the boards used a mixed bag of `TCCR0B`, `TCCR0` and `TCCR0A` with no real delineation lines that could be used for build tags

So far I've split this up at the base chip (attiny85 / atmega328 / atmega2560). That may result in a bit of duplication.

> Also I seem to be incapable of making the dev branch work on my mac, but not for a lack of trying.

Can you share what went wrong, perhaps in a separate issue? If you have LLVM installed with Homebrew (`brew install llvm@10`) it should be as easy as running `go install`, although unfortunately it doesn't seem to work for everybody.

nrwiersma, 2020-07-20:
> Can you share what went wrong, perhaps in a separate issue? If you have LLVM installed with Homebrew (brew install llvm@10) it should be as easy as running go install, although unfortunately it doesn't seem to work for everybody.

It installs "correctly", I get a binary in my Go bin. And I then run `make gen-device-avr`, cause I am only running AVR at this point. but when I run `tinygo info -target arduino-nano` it tells me `LLVM triple: avr-unknown-unknown` and will not build anything. I dont know how to get further than that.

aykevl, 2020-07-20:
> `LLVM triple: avr-unknown-unknown`

This is because the AVR backend is not enabled by default in LLVM 10. It will be in LLVM 11 (which will be released in a few months). If you want to build the dev branch and want to use the AVR backend, you'll need to build LLVM from source, see https://tinygo.org/getting-started/macos/#with-a-self-built-llvm.

nrwiersma, 2020-07-20:
Ok, I will give that a go, thanks.

nrwiersma, 2020-07-21:
An update: I got llvm for build. It took for ever. When running make, it cannot find `libxml2.tbd` even though it is in the MacOS command line tool SDK. I tried making a copy in `llvm-build/lib` but it is not finding it there either. Need to work out what the search path for that is and why it will not find it.

nrwiersma, 2020-07-21:
after trying a bunch of things, added `-v` to the ldflags to see what this thing is doing. And I am now fairly sure the linker is on crack. I get the following output:
```
CGO_CPPFLAGS="-I/Users/nrwiersma/Go/src/github.com/tinygo-org/tinygo/llvm-project/llvm/include -I/Users/nrwiersma/Go/src/github.com/tinygo-org/tinygo/llvm-build/include -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D__STDC_LIMIT_MACROS -I/Users/nrwiersma/Go/src/github.com/tinygo-org/tinygo/llvm-build/tools/clang/include -I/Users/nrwiersma/Go/src/github.com/tinygo-org/tinygo/llvm-project/clang/include -I/Users/nrwiersma/Go/src/github.com/tinygo-org/tinygo/llvm-project/lld/include" CGO_CXXFLAGS="-std=c++14" CGO_LDFLAGS="/Users/nrwiersma/Go/src/github.com/tinygo-org/tinygo/llvm-build/lib/libclang.a -std=c++14 -v -L/Users/nrwiersma/Go/src/github.com/tinygo-org/tinygo/llvm-build/lib  -lclangAnalysis -lclangARCMigrate -lclangAST -lclangASTMatchers -lclangBasic -lclangCodeGen -lclangCrossTU -lclangDriver -lclangDynamicASTMatchers -lclangEdit -lclangFormat -lclangFrontend -lclangFrontendTool -lclangHandleCXX -lclangHandleLLVM -lclangIndex -lclangLex -lclangParse -lclangRewrite -lclangRewriteFrontend -lclangSema -lclangSerialization -lclangStaticAnalyzerCheckers -lclangStaticAnalyzerCore -lclangStaticAnalyzerFrontend -lclangTooling -lclangToolingASTDiff -lclangToolingCore -lclangToolingInclusions  -lstdc++  -llldCOFF -llldCommon -llldCore -llldDriver -llldELF -llldMachO -llldMinGW -llldReaderWriter -llldWasm -llldYAML  -L/Users/nrwiersma/Go/src/github.com/tinygo-org/tinygo/llvm-build/lib -Wl,-search_paths_first -Wl,-headerpad_max_install_names -lLLVMOption -lLLVMMCJIT -lLLVMLTO -lLLVMPasses -lLLVMObjCARCOpts -lLLVMExtensions -lLLVMInterpreter -lLLVMFrontendOpenMP -lLLVMExecutionEngine -lLLVMRuntimeDyld -lLLVMCoverage -lLLVMCoroutines -lLLVMipo -lLLVMInstrumentation -lLLVMVectorize -lLLVMLinker -lLLVMIRReader -lLLVMAsmParser -lLLVMAVRDisassembler -lLLVMAVRCodeGen -lLLVMAVRAsmParser -lLLVMAVRDesc -lLLVMAVRInfo -lLLVMWebAssemblyDisassembler -lLLVMWebAssemblyCodeGen -lLLVMWebAssemblyDesc -lLLVMWebAssemblyAsmParser -lLLVMWebAssemblyInfo -lLLVMRISCVDisassembler -lLLVMRISCVCodeGen -lLLVMRISCVAsmParser -lLLVMRISCVDesc -lLLVMRISCVUtils -lLLVMRISCVInfo -lLLVMAArch64Disassembler -lLLVMAArch64CodeGen -lLLVMAArch64AsmParser -lLLVMAArch64Desc -lLLVMAArch64Utils -lLLVMAArch64Info -lLLVMARMDisassembler -lLLVMARMCodeGen -lLLVMARMAsmParser -lLLVMARMDesc -lLLVMARMUtils -lLLVMARMInfo -lLLVMX86Disassembler -lLLVMX86AsmParser -lLLVMX86CodeGen -lLLVMCFGuard -lLLVMGlobalISel -lLLVMSelectionDAG -lLLVMAsmPrinter -lLLVMDebugInfoDWARF -lLLVMCodeGen -lLLVMTarget -lLLVMScalarOpts -lLLVMInstCombine -lLLVMAggressiveInstCombine -lLLVMTransformUtils -lLLVMBitWriter -lLLVMAnalysis -lLLVMProfileData -lLLVMX86Desc -lLLVMObject -lLLVMTextAPI -lLLVMMCParser -lLLVMBitReader -lLLVMCore -lLLVMRemarks -lLLVMBitstreamReader -lLLVMMCDisassembler -lLLVMMC -lLLVMDebugInfoCodeView -lLLVMDebugInfoMSF -lLLVMBinaryFormat -lLLVMX86Utils -lLLVMX86Info -lLLVMSupport -lLLVMDemangle -lm -llibxml2.tbd -lstdc++ " go build -o build/tinygo -tags byollvm .
# runtime/cgo
Apple clang version 11.0.0 (clang-1100.0.33.8)
Target: x86_64-apple-darwin19.6.0
Thread model: posix
InstalledDir: /Library/Developer/CommandLineTools/usr/bin
 "/Library/Developer/CommandLineTools/usr/bin/ld" -demangle -lto_library /Library/Developer/CommandLineTools/usr/lib/libLTO.dylib -dynamic -arch x86_64 -macosx_version_min 10.15.0 -syslibroot /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk -o $WORK/b059/_cgo_.o -L/Users/nrwiersma/Go/src/github.com/tinygo-org/tinygo/llvm-build/lib -L/Users/nrwiersma/Go/src/github.com/tinygo-org/tinygo/llvm-build/lib $WORK/b059/_cgo_main.o $WORK/b059/_x001.o $WORK/b059/_x002.o $WORK/b059/_x003.o $WORK/b059/_x004.o $WORK/b059/_x005.o $WORK/b059/_x006.o $WORK/b059/_x007.o $WORK/b059/_x008.o $WORK/b059/_x009.o /Users/nrwiersma/Go/src/github.com/tinygo-org/tinygo/llvm-build/lib/libclang.a -lclangAnalysis -lclangARCMigrate -lclangAST -lclangASTMatchers -lclangBasic -lclangCodeGen -lclangCrossTU -lclangDriver -lclangDynamicASTMatchers -lclangEdit -lclangFormat -lclangFrontend -lclangFrontendTool -lclangHandleCXX -lclangHandleLLVM -lclangIndex -lclangLex -lclangParse -lclangRewrite -lclangRewriteFrontend -lclangSema -lclangSerialization -lclangStaticAnalyzerCheckers -lclangStaticAnalyzerCore -lclangStaticAnalyzerFrontend -lclangTooling -lclangToolingASTDiff -lclangToolingCore -lclangToolingInclusions -lc++ -llldCOFF -llldCommon -llldCore -llldDriver -llldELF -llldMachO -llldMinGW -llldReaderWriter -llldWasm -llldYAML -search_paths_first -headerpad_max_install_names -lLLVMOption -lLLVMMCJIT -lLLVMLTO -lLLVMPasses -lLLVMObjCARCOpts -lLLVMExtensions -lLLVMInterpreter -lLLVMFrontendOpenMP -lLLVMExecutionEngine -lLLVMRuntimeDyld -lLLVMCoverage -lLLVMCoroutines -lLLVMipo -lLLVMInstrumentation -lLLVMVectorize -lLLVMLinker -lLLVMIRReader -lLLVMAsmParser -lLLVMAVRDisassembler -lLLVMAVRCodeGen -lLLVMAVRAsmParser -lLLVMAVRDesc -lLLVMAVRInfo -lLLVMWebAssemblyDisassembler -lLLVMWebAssemblyCodeGen -lLLVMWebAssemblyDesc -lLLVMWebAssemblyAsmParser -lLLVMWebAssemblyInfo -lLLVMRISCVDisassembler -lLLVMRISCVCodeGen -lLLVMRISCVAsmParser -lLLVMRISCVDesc -lLLVMRISCVUtils -lLLVMRISCVInfo -lLLVMAArch64Disassembler -lLLVMAArch64CodeGen -lLLVMAArch64AsmParser -lLLVMAArch64Desc -lLLVMAArch64Utils -lLLVMAArch64Info -lLLVMARMDisassembler -lLLVMARMCodeGen -lLLVMARMAsmParser -lLLVMARMDesc -lLLVMARMUtils -lLLVMARMInfo -lLLVMX8 [... cut ...] 86Info -lLLVMSupport -lLLVMDemangle -lm -llibxml2.tbd -lc++ -lpthread -L/usr/local/lib -lSystem /Library/Developer/CommandLineTools/usr/lib/clang/11.0.0/lib/darwin/libclang_rt.osx.a
ld: library not found for -llibxml2.tbd
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [tinygo] Error 2
```

So it acknowledges the SDK path under which `libxml2.tbd` is located, but for some reason cannot find it. I now try removing and resinstalling the command line tools.

nrwiersma, 2020-07-21:
And running `-v` on `clang` confirms it is looking in the place the file exists:
```
$ clang -Xlinker -v
@(#)PROGRAM:ld  PROJECT:ld64-556.6
BUILD 17:57:49 Apr 24 2020
configured to support archs: armv6 armv7 armv7s arm64 arm64e arm64_32 i386 x86_64 x86_64h armv6m armv7k armv7m armv7em
Library search paths:
	/usr/local/lib
	/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/lib
Framework search paths:
	/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/
.... snop ....
```

nrwiersma, 2020-07-21:
and removing `--system-libs` lets it build, but not with the same issues with AVR.
```
./build/tinygo info -target arduino-nano
LLVM triple:       avr-unknown-unknown
GOOS:              linux
GOARCH:            arm
build tags:        avr baremetal linux arm atmega328p atmega avr5 arduino_nano tinygo gc.conservative scheduler.none
garbage collector: conservative
scheduler:         none
```

closed 2020-07-23 by nrwiersma

aykevl, 2020-07-23:
Interesting, was removing `--system-libs` the only change needed?

nrwiersma, 2020-07-23:
It allowed it to build, but it still didnt work.

aykevl, 2020-07-23:
So you still have the same error, that `avr-unknown-unknown` is not a recognized target triple?

nrwiersma, 2020-07-23:
Yip.

aykevl, 2020-07-23:
Strange, the only reason I can imagine that could happen is when you're somehow still linking against a system/Homebrew LLVM.

nrwiersma, 2020-07-24:
I was building with `make` as described in the docs. Homebrew LLVM was removed before this process. I dont know what I could have done differently to make it work.

aykevl, 2020-07-30:
Weird... Without access to the system I don't think I can debug it further. But I'm glad you managed to work around it and get it to work.