tinygo-org/tinygo #1257: proposal: CPU frequency as an untyped const

nrwiersma, 2020-07-22:
Doing math involving the CPU Frequency is fairly painful as it is a function (it cannot make a new const) and it is typed meaning a lot of typecasting needs to be done. I would propose to add it as in a backwards compatible way:

```go
// CPUFreq is the CPU Frequency in hertz.
const CPUFreq = 16000000

// CPUFrequency returns the current CPU frequency in hertz.
func CPUFrequency() uint32 {
	return CPUFreq
}
```

This would allow use of either, which ever is more convenient for the user.

aykevl, 2020-07-22:
Why does it need to be a constant? Usually `CPUFrequency` will get inlined wherever you're calling it and then optimized away. Unfortunately some MCUs allow changing the CPU frequency at runtime so that's why it's a function and not a constant (see https://github.com/tinygo-org/tinygo/pull/784).

nrwiersma, 2020-07-22:
Fair enough. It is not that it needed to be a constant, but that it was untyped, causing far less type conversions when doing math across types. But if there are use cases for it to change then this will not work.

Thanks.

closed 2020-07-22 by nrwiersma