tinygo-org/tinygo #1258: C++ asm in tinygo

nrwiersma, 2020-07-22:
It is unclear how to go from C++ asm statements, including operand constants, to tinygo `AsmFull`.
I tried the example below, but doesnt seem to have the result in llvm I was looking for:
```cpp
__asm__ volatile (
	"1: sbiw %0,1" "\n\t"
	"brne 1b"
	: "=w" (count)
	: "0" (count)
);
```

```go
avr.AsmFull(`
1: sbiw {count},1
   brne 1b
`, map[string]interface{}{
	"count": &count,
})
```

Is there something I have missed here, or a reference I should be using?

aykevl, 2020-07-22:
TinyGo does not support full GCC-style assembly macros. What is your use case for inline assembly?

nrwiersma, 2020-07-22:
I am trying to get fairly accurate timings for specific but variable delays. I need this for an RF433 library when the pulse lengths need to be within a decent margin for it to work. The code above is a small part of a asm delay loop, other asm math needs to be done before this, but I cannot get the same result from C++ to tinygo.

jaddr2line, 2020-07-22:
Do you need the register as an output? If you are just passing it as a parameter and not an output:
```
avr.AsmFull(`
1: sbiw {count},1
   brne 1b
`, map[string]interface{}{
	"count": count,
})
```

Currently output in inline assembly works with a `{}` placeholder which is the destination register. The value of that register is returned by `AsmFull`.

nrwiersma, 2020-07-22:
the `=w` on AVR means that a specific set of registers is used, which ever is free. So it is using `count` as both input and output acting on a specific register.

I tried it without the pointer to count and it didnt like it. Not sure why.

closed 2021-07-09 by nrwiersma