tinygo-org/tinygo #1262: Dev Branch / 0.14 - compiler Panics    "panic: interp: load from a bitcast"

markfarnan, 2020-07-23:
Dev Branch -  Commit:  50a677e9b791bbf34eaa9effca231a036a284ce8
Target:  WASM

Compiles without error in : 0.13

panic: interp: load from a bitcast
```

goroutine 1 [running]:
github.com/tinygo-org/tinygo/interp.(*LocalValue).Load(0xc00c7c2250, 0xc00c7c2250)
        /home/dev/external/tinygo/interp/values.go:61 +0x220
github.com/tinygo-org/tinygo/interp.(*LocalValue).Load(0xc00c7c2240, 0x7f3a0000001d)
        /home/dev/external/tinygo/interp/values.go:58 +0x12f
github.com/tinygo-org/tinygo/interp.(*frame).evalBasicBlock(0xc0065c4990, 0x7f3aa087a9d0, 0x0, 0xc00c7c12a0, 0xc, 0xc000080000, 0x7f3b0084a108, 0x0, 0x7f3a00000000, 0x0, ...)
        /home/dev/external/tinygo/interp/frame.go:98 +0xd684
github.com/tinygo-org/tinygo/interp.(*evalPackage).function(0xc0065c74e0, 0x7f3ac807d2f8, 0xc00920d740, 0x3, 0x4, 0xc00c7c12a0, 0xc, 0x4, 0x0, 0x0)
        /home/dev/external/tinygo/interp/interp.go:113 +0x238
github.com/tinygo-org/tinygo/interp.(*frame).evalBasicBlock(0xc0065c5750, 0x7f3acab22b60, 0x0, 0xc00c7bbe58, 0x8, 0xc000080000, 0x7f3b0084a108, 0x0, 0x7f3a00000000, 0x0, ...)
        /home/dev/external/tinygo/interp/frame.go:570 +0x6463
github.com/tinygo-org/tinygo/interp.(*evalPackage).function(0xc0065c74e0, 0x7f3ac8043c08, 0xc009f46100, 0xa, 0x10, 0xc00c7bbe58, 0x8, 0x10, 0x0, 0x0)
        /home/dev/external/tinygo/interp/interp.go:113 +0x238
github.com/tinygo-org/tinygo/interp.(*frame).evalBasicBlock(0xc0065c6510, 0x7f3acbec4a80, 0x0, 0x4d145f2, 0x4, 0xc0065c64d8, 0xcda65d, 0x7f3acbec4a80, 0x0, 0x0, ...)
        /home/dev/external/tinygo/interp/frame.go:570 +0x6463
github.com/tinygo-org/tinygo/interp.(*evalPackage).function(0xc0065c74e0, 0x7f3ac80680c8, 0xc0087ae100, 0x2, 0x2, 0x4d145f2, 0x4, 0x2, 0xc00c328ce0, 0x0)
        /home/dev/external/tinygo/interp/interp.go:113 +0x238
github.com/tinygo-org/tinygo/interp.(*frame).evalBasicBlock(0xc0065c72d0, 0x7f3acbec3060, 0x7f3acbec2cd0, 0x0, 0x0, 0x0, 0x0, 0xc00c78f370, 0x1, 0x1, ...)
        /home/dev/external/tinygo/interp/frame.go:570 +0x6463
github.com/tinygo-org/tinygo/interp.(*evalPackage).function(0xc0065c74e0, 0x7f3ac8067fd8, 0xc0065c74f8, 0x2, 0x2, 0x0, 0x0, 0x0, 0x0, 0x0)
        /home/dev/external/tinygo/interp/interp.go:113 +0x238
github.com/tinygo-org/tinygo/interp.Run(0x6cda0d0, 0x0, 0x0, 0x0)
        /home/dev/external/tinygo/interp/interp.go:85 +0x7fe
github.com/tinygo-org/tinygo/builder.Build(0x7fffa9da81a8, 0x9, 0x7fffa9da8191, 0x9, 0xc000263200, 0xc0065c7ca0, 0x0, 0x0)
        /home/dev/external/tinygo/builder/build.go:48 +0x20d
main.Build(0x7fffa9da81a8, 0x9, 0x7fffa9da8191, 0x9, 0xc0002040e0, 0x0, 0x12)
        /home/dev/external/tinygo/main.go:97 +0xc5
main.main()
        /home/dev/external/tinygo/main.go:864 +0x1af5
```

---
Bisecting the issue between  
Good:  efdb2e852ec79486cf8556c8b53e389f1f66a0f2
Bad:    50a677e9b791bbf34eaa9effca231a036a284ce8

fed433c04649ebeff2bbcf7a126fa36c525430ea is the first bad commit
commit fed433c04649ebeff2bbcf7a126fa36c525430ea
Author: Ayke van Laethem <aykevanlaethem@gmail.com>
Date:   Fri May 15 23:24:41 2020 +0200

    compiler: add support for atomic operations

    This also implements DisableInterrupts/EnableInterrupts for RISC-V, as
    those operations were needed to implement a few libcalls.

:040000 040000 81f88439a1c3d3501f61a93f83a8892eef589b61 12c84dc8af0e3d2e846859feae2ce518cd6f989f M      compiler
:040000 040000 8e137eb37a59a859fbc6fc93ae314968a0f1d20b 060e444dee60950106593eb551a5cdc14fc42971 M      src
:040000 040000 72f6ceb755fe6c83f39c692d138b1f0f9b058754 ca63a44d8254dffd8be7f7e222d01f42b2623c31 M      testdata

---
Whilst Bisecting, a different panic error appeared earlier:  

171d0fe123b006003032d21885e81df7ceb50623 is the first bad commit
commit 171d0fe123b006003032d21885e81df7ceb50623
Author: Jaden Weiss <jaden@jadendw.dev>
Date:   Mon Jan 27 16:22:46 2020 -0500

    implement mutex blocking

:040000 040000 6c13a6e45eb79b0a20e43315bf45de13d9a9a0cd c89a28ffec1580d07c033dce69f2264267187511 M      src

```
panic: trying to make exported function async: runtime.initAll

goroutine 1 [running]:
github.com/tinygo-org/tinygo/transform.(*coroutineLoweringPass).findAsyncFuncs(0xc0078e11e0)
        /home/dev/external/tinygo/transform/coroutines.go:186 +0xa64
github.com/tinygo-org/tinygo/transform.(*coroutineLoweringPass).load(0xc0078e11e0, 0x20, 0x7f2630e17da0)
        /home/dev/external/tinygo/transform/coroutines.go:305 +0x36c
github.com/tinygo-org/tinygo/transform.LowerCoroutines(0x7e170d0, 0x7e17001, 0x0, 0x0)
        /home/dev/external/tinygo/transform/coroutines.go:70 +0x15f
github.com/tinygo-org/tinygo/transform.Optimize(0x7e170d0, 0xc00023ab10, 0x2, 0x2, 0x5, 0x0, 0x0, 0x0)
        /home/dev/external/tinygo/transform/optimizer.go:120 +0xeb9
github.com/tinygo-org/tinygo/builder.Build(0x7ffef1b041a8, 0x9, 0x7ffef1b04191, 0x9, 0xc00023ab10, 0xc0078e1ca0, 0x0, 0x0)
        /home/dev/external/tinygo/builder/build.go:85 +0x2b55
main.Build(0x7ffef1b041a8, 0x9, 0x7ffef1b04191, 0x9, 0xc0001c00e0, 0x0, 0x12)
        /home/dev/external/tinygo/main.go:97 +0xc5
main.main()
        /home/dev/external/tinygo/main.go:864 +0x1af5
```

aykevl, 2020-07-23:
Can you also share the code that leads to this panic, so we can reproduce this issue?

markfarnan, 2020-07-23:
Code to reproduce: 

Repo:  https://github.com/markfarnan/go-canvas
Branch:   TinyGo-CompileFix
Directory:  ~/go/src/github.com/markfarnan/go-canvas/demo2

Build Cmd:  ~/external/tinygo/build/tinygo  build -o main.wasm -target wasm ./main.go

markfarnan, 2020-07-23:
Pulled PR https://github.com/tinygo-org/tinygo/pull/1263 as requested.  New Error:

```
# image/png
/usr/local/go/src/sync/atomic/value.go:30:20: interp: load from a bitcast

traceback:
/usr/local/go/src/image/format.go:39:34:
  %8 = call %runtime._interface @"(*sync/atomic.Value).Load"(%"sync/atomic.Value"* @image.atomicFormats, i8* undef, i8* undef), !dbg !2386
/usr/local/go/src/image/png/reader.go:1031:22:
  call void @image.RegisterFormat(i8* getelementptr inbounds ([3 x i8], [3 x i8]* @"image/png.init#1$string", i32 0, i32 0), i32 3, i8* getelementptr inbounds ([8 x i8], [8 x i8]* @"image/png.init#1$string.1087", i32 0, i32 0), i32 8, i8* undef, i32 ptrtoint (%runtime.funcValueWithSignature* @"image/png.Decode$withSignature" to i32), i8* undef, i32 ptrtoint (%runtime.funcValueWithSignature* @"image/png.DecodeConfig$withSignature" to i32), i8* undef, i8* undef), !dbg !2367
image/png/<init>:
  call void @"image/png.init#1"(i8* undef, i8* undef), !dbg !2369
```

aykevl, 2020-07-24:
The following code is enough to reproduce:

```go
package main

import _ "image/png"

func main() {}
```

aykevl, 2020-07-30:
I think I have a complete fix, see #1265. @markfarnan can you try that PR to see whether it fixes your problem?

markfarnan, 2020-07-31:
Confirm I can now compile the code which was failing previously.   

I can't confirm 100% as now having a run-time error, which I will investigate further.  
   - App compiled with TinyGo fails,  with Go 1.14 works fine, only change is different wasm_exec.js for the relevant version.  

Uncaught (in promise) RuntimeError: unreachable
    at runtime.runtimePanic (http://localhost:8080/main.wasm:wasm-function[86]:0x5ff5)
    at runtime.alloc (http://localhost:8080/main.wasm:wasm-function[29]:0x1c4d)
    at command-line-arguments.main (http://localhost:8080/main.wasm:wasm-function[207]:0x16519)
    at runtime.run$1.resume (http://localhost:8080/main.wasm:wasm-function[124]:0x8a62)
    at runtime.scheduler (http://localhost:8080/main.wasm:wasm-function[81]:0x5d59)
    at _start (http://localhost:8080/main.wasm:wasm-function[79]:0x5853)
    at global.Go.run (http://localhost:8080/wasm_exec.js:469:24)
    at http://localhost:8080/:38:7

runtime.runtimePanic | @ | main.wasm:formatted:896
-- | -- | --
  | runtime.alloc | @ | main.wasm:formatted:149
  | command-line-arguments.main | @ | main.wasm:formatted:3735
  | runtime.run$1.resume | @ | main.wasm:formatted:1344
  | runtime.scheduler | @ | main.wasm:formatted:870
  | _start | @ | main.wasm:formatted:813
  | run | @ | wasm_exec.js:469
  | (anonymous) | @ | (index):38
  | Promise.then (async) |   |  
  | (anonymous) | @ | (index):37

jaddr2line, 2020-07-31:
Huh. It looks like we somehow ended up with a UB code path that was actually reachable.

jaddr2line, 2020-07-31:
Can you please provide the panic message?

markfarnan, 2020-07-31:
panic: runtime error: out of memory   wasm_exec.js:267

As discussed on Slack,  was allocating more than the 1mb heap for the Canvas frame buffer.  

-heap-size=16m   sorted it out, and canvas demo is now plotting etc. 

Calling this part  working, 

Will test again once merged for Dev 0.14 etc. 

Thanks.

deadprogram, 2020-08-07:
This was released with v0.14.0 so now closing. Please reopen if needed. Thanks!

closed 2020-08-07 by deadprogram