tinygo-org/tinygo #1266: Printf causes random hard faults

timboldt, 2020-07-25:
I discovered that trying to print floating point numbers results in random hard faults, on a Feather M0.

For example, this was problematic:

tinyfont.WriteLine(&display, &proggy.TinySZ8pt7b, 0, 6, []byte(
			fmt.Sprintf("%.1f'C  %.2f V",
				temperature,
				batteryVoltage)),
			color.RGBA{255, 255, 255, 255})

Context:
https://github.com/timboldt/epaper-display/blob/e5169a18099b5f1563cbaab0452d9a360fa76351/main.go
TinyGo version: tinygo version 0.13.1 linux/amd64 (using go version go1.14.3 and LLVM version 10.0.1)
Hardware: Adafruit Feather M0 Express

jaddr2line, 2020-07-25:
I think this might be fixed by #1105 which is currently on `dev`? I will look into this later today.

timboldt, 2020-07-25:
I had thought so at first too, but the Feather M0 doesn't have an FPU.

deadprogram, 2020-09-12:
Is this still an issue with the current `dev` branch?

bgould, 2023-02-01:
I tested the sample program using Feather M0, and the issue appears to be fixed.  Program runs indefinitely without panic.  Closing out as fixed.  Thank you!

closed 2023-02-01 by bgould