tinygo-org/tinygo #1267: Building programs for i686-unknown-linux-gnu spits out "undeclared name" errors

foxcpp, 2020-07-26:
https://tinygo.org/linux/ says:
>TinyGo also lets you compile programs for Linux systems, both **32-bit** and 64-bit, on **both x86** and ARM architectures.

## What I do 

```
package main

func main() {
	fact := 1
	for i := 1; i < 10; i++ {
		fact *= 1
	}
}
```

I try building it for 32-bit x86 linux using:
```
tinygo build -target i686-unknown-linux-gnu -o test.elf
```

## What I expect

ELF binary produced and linked.

## What I get

```
# runtime
../../../usr/lib/tinygo/src/runtime/math.go:60:5: undeclared name: GOARCH
../../../usr/lib/tinygo/src/runtime/math.go:60:26: undeclared name: GOARCH
../../../usr/lib/tinygo/src/runtime/math.go:116:5: undeclared name: GOARCH
../../../usr/lib/tinygo/src/runtime/math.go:116:26: undeclared name: GOARCH
../../../usr/lib/tinygo/src/runtime/math.go:172:5: undeclared name: GOARCH
../../../usr/lib/tinygo/src/runtime/math.go:172:26: undeclared name: GOARCH
../../../usr/lib/tinygo/src/runtime/math.go:186:5: undeclared name: GOARCH
../../../usr/lib/tinygo/src/runtime/math.go:186:26: undeclared name: GOARCH
../../../usr/lib/tinygo/src/runtime/math.go:236:5: undeclared name: GOARCH
../../../usr/lib/tinygo/src/runtime/math.go:236:24: undeclared name: GOARCH
../../../usr/lib/tinygo/src/runtime/math.go:236:45: undeclared name: GOARCH
../../../usr/lib/tinygo/src/runtime/math.go:262:5: undeclared name: GOARCH
../../../usr/lib/tinygo/src/runtime/math.go:262:26: undeclared name: GOARCH
../../../usr/lib/tinygo/src/runtime/print.go:19:5: undeclared name: TargetBits
../../../usr/lib/tinygo/src/runtime/print.go:31:5: undeclared name: TargetBits
../../../usr/lib/tinygo/src/runtime/scheduler_coroutines.go:8:9: undeclared name: getCurrentStackPointer
```

Footnote: Building for x86_64-unknown-linux-gnu works correctly.

## tinygo version

```
tinygo version 0.13.1 linux/amd64 (using go version go1.14.4 and LLVM version 10.0.0)
```

```
GOOS="linux"
GOARCH="amd64"
GOROOT="/usr/lib/go"
GOPATH="/home/foxcpp/.go"
GOCACHE="/home/foxcpp/.cache/tinygo"
CGO_ENABLED="1"
TINYGOROOT="/usr/lib/tinygo"
```

aykevl, 2020-07-26:
For this you should use `GOARCH`. The documentation doesn't say this yet but it will be fixed with the next release (which also includes updated documentation).

    GOARCH=386 tinygo build -o test.elf

closed 2020-07-26 by aykevl

foxcpp, 2020-07-26:
Yeah, that works... Thanks for information!