tinygo-org/tinygo #1270: bluepill flashing not working

ski7777, 2020-07-27:
I'm very new to tinygo and I wanted try it on my bluepill board.
Sadly flashing does not work. I tested it using a cheap china ST-LINK v2 and the Nucleo ST-Link (broken apart from nucleo itself).
This is my Code:
```go
package main

import (
	"machine"
	"time"
)

func main() {
	led := machine.LED
	led.Configure(machine.PinConfig{Mode: machine.PinOutput})
	for {
		led.Low()
		time.Sleep(time.Millisecond * 1000)

		led.High()
		time.Sleep(time.Millisecond * 1000)
	}
}
```

This is the shell output:
```
tinygo flash -target bluepill main.go
Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
adapter speed: 1000 kHz
adapter_nsrst_delay: 100
none separate
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : clock speed 950 kHz
Error: open failed
in procedure 'program' 
in procedure 'init' called at file "embedded:startup.tcl", line 506
in procedure 'ocd_bouncer'
** OpenOCD init failed **
shutdown command invoked

error: failed to flash /tmp/tinygo361065343/main.hex: exit status 1
```

Is there any magic I forgot? Help is really appreciated.

Thanks
Raphael

aykevl, 2020-07-28:
It appears that the ST-Link v2 wasn't properly connected, otherwise you'd see a message like `Target voltage: 3.249768`. There might be a Linux permission problem so that OpenOCD cannot access the programmer, perhaps you could check dmesg?

ski7777, 2020-07-28:
Do I need special udev configuration?

alankrantas, 2020-07-28:
Just tried it on my RPi 3 and found a solution. You'll need to grant permission for openOCD:

```
sudo nano /etc/udev/rules.d/98-stlink.rules
```

and add

```
SUBSYSTEM=="usb", ATTR{idVendor}=="0483", ATTR{idProduct}=="374b", MODE="666"
SUBSYSTEM=="usb", ATTR{idVendor}=="0483", ATTR{idProduct}=="3748", MODE="666"
SUBSYSTEM=="usb", ATTR{idVendor}=="0483", ATTR{idProduct}=="3744", MODE="666"
```

in the file.

ski7777, 2020-07-28:
I applied this. Next error:
```
Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
adapter speed: 1000 kHz
adapter_nsrst_delay: 100
none separate
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : clock speed 950 kHz
Info : STLINK v2 JTAG v28 API v2 SWIM v7 VID 0x0483 PID 0x3748
Info : using stlink api v2
Info : Target voltage: 3.244444
Warn : UNEXPECTED idcode: 0x2ba01477
Error: expected 1 of 1: 0x1ba01477
in procedure 'program' 
in procedure 'init' called at file "embedded:startup.tcl", line 506
in procedure 'ocd_bouncer'
** OpenOCD init failed **
shutdown command invoked

error: failed to flash /tmp/tinygo060123624/main.hex: exit status 1
```

How do I need to set the BOOT jumpers? Do I need to connect the RST line?

Thanks
Raphael

alankrantas, 2020-07-28:
Maybe this would help?
https://community.platformio.org/t/cannot-upload-to-stm32-bluepill-board-over-stlink-v2/3492/13

ski7777, 2020-07-28:
This seems to go in the right direction but how can I do this in tinygo?

alankrantas, 2020-07-29:
You simply modify openOCD's config file on your own system.

ski7777, 2020-07-29:
But where can I find this file?

alankrantas, 2020-07-29:
I installed openOCD in my working laptop's VM Ubuntu, and it's at **/usr/share/openocd/scripts/target/stm32f1x.cfg**.

I'll take a look at my RPi when I get home.

ski7777, 2020-07-29:
Now it works but I think that should not be the way to go. A global fix would be better :)

kenbell, 2021-07-05:
This is fixed given there's a `bluepill-clone` target? (which worked well enough for me on a cloned bluepill)

I'll close in a few days unless someone thinks there's a reason to keep this open.

kenbell, 2021-07-09:
Closing since there's the `bluepill-clone` target

closed 2021-07-09 by kenbell