tinygo-org/tinygo #1272: Error using goroutines: attempted to start a goroutine without a scheduler

willmendil, 2020-07-30:
Hi, 

when I want to do a simple goroutine, (looking for button press) I get this compilation error
```
main.go:596:2: attempted to start a goroutine without a scheduler
```

I see that goroutines are supported on arduino (in my case nano). What am I missing? what is a scheduler?

alankrantas, 2020-07-30:
Scheduler is default off on 8-bit AVR boards I think. It's the Golang mechanism to run goroutines.

You have to enable it like this:

```
tinygo flash -scheduler coroutines -target xxx
```

willmendil, 2020-07-30:
oh, simple enough. I did not see that in the documentation. 
(side question, i saw you talked about servo library, on another issue. Do you have info on that?)

jaddr2line, 2020-07-30:
AVR does not support coroutines. You need `-scheduler tasks`.

closed 2020-07-30 by willmendil

alankrantas, 2020-07-30:
Interesting, because I can run the blinky2 example (which use goroutines) on my Uno with the -scheduler flag. I'll try the other one later.

Servo control involves PWM frequency setting and it turned out to be difficult to implement that function across different hardwares. So no luck for now.

jaddr2line, 2020-07-30:
Hmm. . . maybe LLVM fixed the bug, or maybe it is just somehow working anyway.

alankrantas, 2020-07-30:
Yeah... just tried it with two LEDs. Both ```-scheduler coroutines``` and ```-scheduler tasks``` compiles and runs. However the timing is terrible, very inaccurate. :p