tinygo-org/tinygo #1273: Bluepill: fix .hex file path for openOCD on Windows

alankrantas, 2020-07-31:
```
PS C:\Users\Admin> tinygo flash -target bluepill examples/blinky1
Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
adapter speed: 1000 kHz
adapter_nsrst_delay: 100
none separate
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : clock speed 950 kHz
Info : STLINK v2 JTAG v34 API v2 SWIM v7 VID 0x0483 PID 0x3748
Info : using stlink api v2
Info : Target voltage: 3.250895
Info : stm32f1x.cpu: hardware has 6 breakpoints, 4 watchpoints
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x0800069c msp: 0x20000800
** Programming Started **
auto erase enabled
Error: Invalid command argument
**image.base_address option value ('inygo008418063main.hex') is not valid**
** Programming Failed **
shutdown command invoked
```

I think the problems is (quote) "OpenOCD uses Tcl and a backslash is an escape char", so you'll need to use double backslash in the path.

For now I can build a .hex and flash it via the ST-Link Utility, and now I can do it on my RPi, so this is not an urgent issue.

Also, [this](https://freddiechopin.info/en/download/category/4-openocd) is where I found the openOCD with the executable file in it. Somehow the official release does not include it.

sago35, 2020-08-17:
You are right.

I was able to reproduce it too.
I think it's easy to fix.

I want to use openocd on windows too and will fix it.

NG: single backslash

```
C:\tinygo>openocd -f flash_elf.tcl -c "program .\app.hex reset exit"
Open On-Chip Debugger 0.10.0+dev-g5c9350d2-dirty (2020-08-16-22:05)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "swd". To override use 'transport select <transport>'.
force hard breakpoints
Info : Using CMSIS-DAPv2 interface with VID:PID=0x6666:0x6666, serial=123456
Info : CMSIS-DAP: SWD  Supported
Info : CMSIS-DAP: FW Version = v0.1
Info : CMSIS-DAP: Serial# = 123456
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : SWCLK/TCK = 1 SWDIO/TMS = 1 TDI = 1 TDO = 1 nTRST = 0 nRESET = 1
Info : CMSIS-DAP: Interface ready
Info : clock speed 3200 kHz
Info : SWD DPIDR 0x1ba01477
Info : stm32f1x.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : starting gdb server for stm32f1x.cpu on 3333
Info : Listening on port 3333 for gdb connections
flash_elf
target halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x080006f8 msp: 0x20000800
** Programming Started **
Error: couldn't open .pp.hex
** Programming Failed **
shutdown command invoked
```

OK: toSlash()

```
C:\tinygo>openocd -f flash_elf.tcl -c "program C:/tinygo/app100.hex reset exit"
Open On-Chip Debugger 0.10.0+dev-g5c9350d2-dirty (2020-08-16-22:05)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "swd". To override use 'transport select <transport>'.
force hard breakpoints
Info : Using CMSIS-DAPv2 interface with VID:PID=0x6666:0x6666, serial=123456
Info : CMSIS-DAP: SWD  Supported
Info : CMSIS-DAP: FW Version = v0.1
Info : CMSIS-DAP: Serial# = 123456
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : SWCLK/TCK = 1 SWDIO/TMS = 1 TDI = 1 TDO = 1 nTRST = 0 nRESET = 1
Info : CMSIS-DAP: Interface ready
Info : clock speed 3200 kHz
Info : SWD DPIDR 0x1ba01477
Info : stm32f1x.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : starting gdb server for stm32f1x.cpu on 3333
Info : Listening on port 3333 for gdb connections
flash_elf
target halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x080006f8 msp: 0x20000800
** Programming Started **
Info : device id = 0x20036410
Info : ignoring flash probed value, using configured bank size
Info : flash size = 128kbytes
** Programming Finished **
** Resetting Target **
shutdown command invoked
```

OK: double backslash

```
C:\tinygo>openocd -f flash_elf.tcl -c "program C:\\tinygo\\app100.hex reset exit"
Open On-Chip Debugger 0.10.0+dev-g5c9350d2-dirty (2020-08-16-22:05)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "swd". To override use 'transport select <transport>'.
force hard breakpoints
Info : Using CMSIS-DAPv2 interface with VID:PID=0x6666:0x6666, serial=123456
Info : CMSIS-DAP: SWD  Supported
Info : CMSIS-DAP: FW Version = v0.1
Info : CMSIS-DAP: Serial# = 123456
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : SWCLK/TCK = 1 SWDIO/TMS = 1 TDI = 1 TDO = 1 nTRST = 0 nRESET = 1
Info : CMSIS-DAP: Interface ready
Info : clock speed 3200 kHz
Info : SWD DPIDR 0x1ba01477
Info : stm32f1x.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : starting gdb server for stm32f1x.cpu on 3333
Info : Listening on port 3333 for gdb connections
flash_elf
target halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x080006f8 msp: 0x20000800
** Programming Started **
Info : device id = 0x20036410
Info : ignoring flash probed value, using configured bank size
Info : flash size = 128kbytes
** Programming Finished **
** Resetting Target **
shutdown command invoked
```

sago35, 2020-08-17:
not correct: Error: couldn't open C:UsersTAKASA~1.MASAppDataLocalTemp        inygo987121619main.hex
correct: C:\Users\TAKASA~1.MAS\AppData\Local\Temp\tinygo987121619\main.hex

NG: tinygo flash

```
C:\tinygo>tinygo flash -target bluepill-cmsis-dap .\src\examples\blinky1
Open On-Chip Debugger 0.10.0+dev-g5c9350d2-dirty (2020-08-16-22:05)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "swd". To override use 'transport select <transport>'.
Info : Using CMSIS-DAPv2 interface with VID:PID=0x6666:0x6666, serial=123456
Info : CMSIS-DAP: SWD  Supported
Info : CMSIS-DAP: FW Version = v0.1
Info : CMSIS-DAP: Serial# = 123456
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : SWCLK/TCK = 1 SWDIO/TMS = 1 TDI = 1 TDO = 1 nTRST = 0 nRESET = 1
Info : CMSIS-DAP: Interface ready
Info : clock speed 1000 kHz
Info : SWD DPIDR 0x1ba01477
Info : stm32f1x.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : starting gdb server for stm32f1x.cpu on 3333
Info : Listening on port 3333 for gdb connections
target halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x080006f8 msp: 0x20000800
** Programming Started **
Error: couldn't open C:UsersTAKASA~1.MASAppDataLocalTemp        inygo987121619main.hex
** Programming Failed **
shutdown command invoked

error: failed to flash C:\Users\TAKASA~1.MAS\AppData\Local\Temp\tinygo987121619\main.hex: exit status 1
```

sago35, 2020-08-27:
I think we've solved the problem.
If you want to use the resolved version, you can download it below.

https://dev.azure.com/tinygo/tinygo/_build/results?buildId=1994&view=artifacts&type=publishedArtifacts

alankrantas, 2020-08-27:
@sago35 it works, thank you! I'll close this now.

closed 2020-08-27 by alankrantas