tinygo-org/tinygo #1278: Flashing on an adafruit nrf52 board causes an error message to appear

sago35, 2020-08-03:
When I try to do tinygo flash with feather-nrf52840, I get an error message.
tinygo flash seems to succeed.

The reset appears to be running immediately after the following.
https://github.com/tinygo-org/tinygo/blob/dev/main.go#L82

It's the same on both ubuntu and windows.Is this the same for everyone?
We might want to include a note of caution in your documentation.

```
# ubuntu 20.04
$ tinygo flash --target feather-nrf52840 --size short ./src/examples/blinky1
   code    data     bss |   flash     ram
   6256      12    4064 |    6268    4076
error: failed to flash /tmp/tinygo749323199/main.uf2: rename /media/sago35/FTHR840BOOT/flash.uf2.tmp /media/sago35/FTHR840BOOT/flash.uf2: no such file or directory
```

```
# windows10 pro
$ .\bin\tinygo.exe flash --target feather-nrf52840 --size short .\src\examples\blinky1
   code    data     bss |   flash     ram
   6856      12    4064 |    6868    4076
error: failed to flash C:\tinygo-pr\tinygo0.14.0dev_1afe528\temp\tinygo458391831\main.uf2: rename D:\/flash.uf2.tmp D:\/flash.uf2: A device which does not exist was specified.
```

sago35, 2020-08-03:
> Benjamin Gould  19:57
I'm able to reproduce the same problem using `circuitplay-bluefruit` and `clue_alpha` targets as well

sago35, 2020-08-03:
https://github.com/adafruit/Adafruit_nRF52_Bootloader

sago35, 2020-08-03:
I don't know if this is relevant or not.
https://github.com/hathach/tinyusb/issues/375
https://github.com/adafruit/Adafruit_nRF52_Bootloader/issues/120

bgould, 2020-08-04:
FWIW this is working ok on my Debian laptop, but reproducible using the same board on my Windows computer as well as an Ubuntu VM running in VirtualBox on that same Windows computer.  These are the bootloader details for that board:
```
UF2 Bootloader 0.2.6 lib/nrfx (v1.1.0-1-g096e770) lib/tinyusb (legacy-525-ga1c59649) s140 6.1.1                 
Model: Adafruit Feather nRF52840 Express                                        
Board-ID: NRF52-Bluefruit-v0                                                    
Bootloader: s140 6.1.1                                                          
Date: Dec 21 2018
```

sago35, 2020-08-25:
It has been fixed and will be closed.

deadprogram, 2020-12-28:
Something is wrong again with the same symptoms as before. Here is how to reproduce:

Double click into bootloader on board, then flash blinky1:

```
tinygo build -o /media/ron/ITSY840BOOT/flash.uf2 -target itsybitsy-nrf52840 examples/blinky1
```

board will flash and restart, blinking as expected.

Now you can use the normal `tinygo flash` command with the blinky1 program:

```
tinygo flash -target itsybitsy-nrf52840 examples/blinky1
```

board flashes and restarts as expected. You can even run the above command several times.

However, if you use any of the programs that use the USB CDC port:

```
tinygo flash -target itsybitsy-nrf52840 examples/echo
```

The board flashes but does not restart unless you single press the button on the board after flashing. Once you have restarted the board, it acts normally again until flashing.

deadprogram, 2020-12-28:
https://github.com/tinygo-org/tinygo/pull/1548 should address the root of this issue.

deadprogram, 2020-12-29:
#1548 has been merged so will close again next release.

deadprogram, 2021-03-08:
This has been released with v0.17.0 so now closing. Thanks!

closed 2021-03-08 by deadprogram