tinygo-org/tinygo #1279: Add support for additional openocd commands

supcik, 2020-08-04:
`openocd` accepts additional commands using the `--command` or `-c` option. This is useful, for example, if you want to override some variables. A typical [use case](https://community.platformio.org/t/cannot-upload-to-stm32-bluepill-board-over-stlink-v2/3492/22) is to override the `CPUTAPID` for some [stm32f1 clones](https://community.platformio.org/t/cannot-upload-to-stm32-bluepill-board-over-stlink-v2/3492/13) using the argument `-c set CPUTAPID 0x2ba01477`.

A possible implementation in TinyGo could be done using a new key/value (for example `openocd-commands: ...`) in the `targets/*.json` files. The `targets/bluepill-clone.json` could then be:

```json
{
	"inherits": ["cortex-m"],
	"llvm-target": "armv7m-none-eabi",
	"build-tags": ["bluepill", "stm32f103xx", "stm32"],
	"cflags": [
		"--target=armv7m-none-eabi",
		"-Qunused-arguments"
	],
	"linkerscript": "targets/stm32.ld",
	"extra-files": [
		"src/device/stm32/stm32f103xx.s"
	],
	"flash-method": "openocd",
	"openocd-interface": "stlink-v2",
	"openocd-target": "stm32f1x",
	"openocd-commands": ["set CPUTAPID 0x2ba01477"]
}
```

supcik, 2020-08-04:
This would also solve #1270

supcik, 2020-08-08:
The `.json` file could even be simpler using inheritance:

```json
{
	"inherits": ["bluepill"],
	"openocd-commands": ["set CPUTAPID 0x2ba01477"]
}
```

deadprogram, 2021-03-08:
This has been released with v0.17.0 so now closing. Thanks!

closed 2021-03-08 by deadprogram