tinygo-org/tinygo #1285: Missing info functions from `runtime` and `os`

ribasushi, 2020-08-09:
**TLDR**
`os` is listed as fully implemented, while `runtime` is [missing entirely in the stdlib list](https://tinygo.org/lang-support/stdlib/). In any case some of the constant-returns are not there.

**Version**
`tinygo version 0.14.0 darwin/amd64 (using go version go1.14.7 and LLVM version 10.0.0)`

**Failure**
```
~$ tinygo run tiny.go 
# command-line-arguments
tiny.go:14:11: Version not declared by package runtime
tiny.go:15:11: NumCPU not declared by package runtime
tiny.go:16:11: NumGoroutine not declared by package runtime
tiny.go:17:6: Getpagesize not declared by package os
```

**Expectation**
```
~$ go run tiny.go 
runtime.Version:        go1.14.7
runtime.NumCPU: 8
runtime.NumGoroutine:   1
os.Getpagesize: 4096
```

**Testcase**
```
package main

import (
	"fmt"
	"os"
	"runtime"
)

func main() {
	fmt.Printf("runtime.Version:\t%s\n"+
		"runtime.NumCPU:\t%d\n"+
		"runtime.NumGoroutine:\t%d\n"+
		"os.Getpagesize:\t%d\n",
		runtime.Version(),
		runtime.NumCPU(),
		runtime.NumGoroutine(),
		os.Getpagesize(),
	)
}
```

jaddr2line, 2020-08-09:
- `runtime.NumCPU` - right now this should just return 1
- `runtime.NumGoroutine` - we do not currently track this; maybe we should
- `os.Getpagesize` - most TinyGo targets do not have pages, so I have no idea what this should return
- `runtime.Version` - seems reasonable enough

ribasushi, 2020-08-09:
> runtime.NumGoroutine - we do not currently track this; maybe we should

I use it in certain shutdown scenarios as "did I really exit everything? or are there dangling goroutines?" It can be put behind a build-tag, but having the actual counter exposed is of course much better.

> os.Getpagesize - most TinyGo targets do not have pages, so I have no idea what this should return

It returns an int, so `-1` seems reasonable. Perhaps asking go-core is worthwhile.

aykevl, 2020-10-27:
> **TLDR**
> `os` is listed as fully implemented, while `runtime` is [missing entirely in the stdlib list](https://tinygo.org/lang-support/stdlib/). In any case some of the constant-returns are not there.

No, see the text at the top of the page:

> The following table shows all Go standard library packages and whether they can be imported by TinyGo. If they can’t, you can click the ‘no’ link to jump to the explanation why the package cannot be compiled.
> 
> Note that the fact they can be imported, does not mean that all functions and types in the program can be used. For example, sometimes using some functions or types of the package will still trigger compiler errors.

All this does is try to import the package and see whether that works at all.

ribasushi, 2020-10-28:
@aykevl be that as it may, what can be done about implementing the missing pieces ( see prior discussion )

dgryski, 2023-01-31:
Some of these runutime stubs are easy enough to implement.  Maybe this should be tagged as "good first issue" ?

dgryski, 2024-10-21:
These functions exist now.
```
~/go/src/github.com/dgryski/bug/osinfo $ tinygo run main.go
runtime.Version:	0.34.0-dev
runtime.NumCPU:	1
runtime.NumGoroutine:	1
os.Getpagesize:	4096
~/go/src/github.com/dgryski/bug/osinfo $ tinygo run -target=wasip1 main.go
runtime.Version:	0.34.0-dev
runtime.NumCPU:	1
runtime.NumGoroutine:	1
os.Getpagesize:	65536
```

closed 2024-10-21 by dgryski