tinygo-org/tinygo #1286: ENOTDIR not declared by package syscall

ribasushi, 2020-08-09:
**TLDR**
`path/filepath` is listed as fully implemented, yet it fails compilation.

**Version**
`tinygo version 0.14.0 darwin/amd64 (using go version go1.14.7 and LLVM version 10.0.0)`

**Failure**
```
~$ tinygo run tiny.go 
# path/filepath
/usr/local/go/src/path/filepath/symlink.go:90:24: ENOTDIR not declared by package syscall
```

**Expectation**
```
~$ go run tiny.go 
cur dir:        /Users/{{expected stuff follows here}}
```

**Testcase**
```
package main

import (
	"fmt"
	"path/filepath"
)

func main() {
	cwd, _ := filepath.Abs(".")
	fmt.Printf("cur dir:\t%s\n", cwd)
}
```

ribasushi, 2020-08-10:
Is this a difficult issue to solve? Of all the tickets I filed yesterday, this one is the main blocker, as it prevents compilation of a dependency I can't easily get away from 😿 
I am very willing to try a branch ( `dev` or otherwise ) and I'd even try my hand at fixing this if one gives me a general "area to dig in".

Thanks!

codefromthecrypt, 2022-09-07:
@ribasushi I'm sorry this was frustrating for you. This is now sorted. If you don't mind closing the issue, I'd appreciate it. I'm a consumer of TinyGo and felt maybe closing some of the hundreds of open issues might lead to quicker turnaround for the next person, hence bugging you.

niaow, 2022-11-04:
Fixed by #2268

closed 2022-11-04 by niaow