tinygo-org/tinygo #1287: `os.Stdin.Stat()` panics

ribasushi, 2020-08-09:
**TLDR**
Stat on `Stdin` ( be it a tty, a pipe or a file ) panics

**Version**
`tinygo version 0.14.0 darwin/amd64 (using go version go1.14.7 and LLVM version 10.0.0)`

**Failure**
```
~$ tinygo run tiny.go
panic: runtime error: nil pointer dereference
error: failed to run compiled binary /var/folders/3y/58n01w8x1xz1wv08x6xflq0h0000gn/T/tinygo353964835/main: signal: abort trap

~$ echo foo | tinygo run tiny.go
panic: runtime error: nil pointer dereference
error: failed to run compiled binary /var/folders/3y/58n01w8x1xz1wv08x6xflq0h0000gn/T/tinygo878131135/main: signal: abort trap

~$ <tiny.go tinygo run tiny.go
panic: runtime error: nil pointer dereference
error: failed to run compiled binary /var/folders/3y/58n01w8x1xz1wv08x6xflq0h0000gn/T/tinygo967377983/main: signal: abort trap
```

**Expectation**
```
~$ go run tiny.go 
in perm:        0620

~$ echo foo | go run tiny.go 
in perm:        0660

~$ <tiny.go go run tiny.go 
in perm:        0644
```

**Testcase**
```
package main

import (
	"fmt"
	"os"
)

func main() {
	inStat, _ := os.Stdin.Stat()
	fmt.Printf("in perm:\t0%o\n", inStat.Mode()&07777)
}
```

codefromthecrypt, 2022-09-07:
@ribasushi I noticed this doesn't panic using latest tinygo, though it always seems to report 0666 in tinygo despite go reporting various modes. If you are satisfied with it not panicing can you close this out? Otherwise, mention what's next? Cheers!

dgryski, 2023-02-01:
Note that binaries built with `tinygo` report the permissions correctly.  This bug has to do with how tinygo sets up the sub-processes.

closed 2026-09-19 by deadprogram