tinygo-org/tinygo #1288: Under win10, use tinygo to build wasm to report an error

funjobs, 2020-08-11:
When I try to build the sample code in "USING WEBASSEMBLY", the error is as follows:
`$ tinygo build -o wasm.wasm -target wasm ./app.go
error: symlink D:\tinygo\src\os\env.go C:\Users\zhaoxin\AppData\Local\tinygo\goroot-go1.14.7-bdc2af1295a41498a02f0d99a3cf01f71e2ccaeb0bdebae56c969f113bbaeb7b.tmp5577006791947779410\src\os\env.go: A required privilege is not held by the client.
`

sago35, 2020-08-11:
I cann't reproduce it.
Please show me the source code if possible.

```
$ tinygo build -o wasm.wasm -target wasm .\main\main.go

$ dir wasm.wasm
2020/08/12  07:05            94,202 wasm.wasm

$ type .\main\main.go
package main

import (
        "os"
)

func main() {
        println("Hello world!")
        println(os.Getenv("hello"))
}

$ tinygo version
tinygo version 0.14.0 windows/amd64 (using go version go1.14.3 and LLVM version 10.0.1)
```

funjobs, 2020-08-12:
[image]
OK! That's all.

aykevl, 2020-08-13:
It appears you have TinyGo on a different filesystem than your cache directory. You can work around this issue in one of the following ways:

* Move the TinyGo install directory to your C drive.
* Enable Developer Mode in Windows.

Leaving this open as this can also be fixed in TinyGo directly.

aykevl, 2020-08-13:
@funjobs can you check:

1. That you can still reproduce this issue
2. That this build fixes the issue (when you put it on the same D: drive): https://dev.azure.com/tinygo/tinygo/_build/results?buildId=1961&view=artifacts&type=publishedArtifacts

I hope to have fixed the issue, but if you can please confirm that it works.

funjobs, 2020-08-14:
Great, the compilation is successful! Thank you for your patience.

closed 2020-08-14 by funjobs

aykevl, 2020-08-15:
@funjobs thank you for testing!