tinygo-org/tinygo #1290: Arduino Mega 1280

MrNinso, 2020-08-12:
When I tried to flash: 
````golang
package main

import (
	"machine"
	"time"
)

func main() {
	led := machine.D22

	led.Configure(machine.PinConfig{Mode: machine.PinOutput})

	for {
		println("LOW")
		led.Low()
		time.Sleep(500000000)
		println("HIGH")
		led.High()
		time.Sleep(500000000)
	}
}
````

Using:

````
tinygo flash -target arduino-mega2560 -port /dev/ttyUSB0 blink.go
````
I was getting:

````
	avrdude: Version 6.3, compiled on Jan 28 2020 at 00:00:00
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/etc/avrdude/avrdude.conf"
         User configuration file is "/home/developer/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/ttyUSB0
         Using Programmer              : wiring
         Overriding Baud Rate          : 115200
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_getsync(): timeout communicating with programmer

avrdude done.  Thank you.

error: failed to flash /tmp/tinygo654791811/main.hex: exit status 1
````

but Arduino IDE was working, so I debug it to know how Arduino IDE was uploading the .hex file, and I noticed the Arduino IDE is using:
````
avrdude -c arduino -b 57200 -p atmega1280 -P {port} -U flash:w:{hex} -v -D
````

so, I copyed ``arduino-mega2560.json`` and change the flash-command field. and "worked"

basically after this change I can flash but:

- Some analog in pins are inverted, and some are duplicated   
- I can write in digital pins but if I try to read, return false no matter what

aykevl, 2020-08-13:
What board are you using, the Arduino Mega 1280 or the Arduino Mega 2560? These are two different boards, and only the Arduino Mega 2560 is supported right now.

MrNinso, 2020-08-13:
I'm Using  Arduino Mega 1280 with ATMEGA1280, but  if I am not mistaken the different between Arduino Mega 1280 and Arduino Mega 2560 is the flash memory.

aykevl, 2020-08-13:
Well looks like the _chip_ is indeed very similar (almost identical). Although I should note that the ATmega1280 and ATmega2560 are architecturally incompatible (they use a slightly different variant of the AVR architecture that is not compatible due to the larger flash size). However, LLVM doesn't yet take that into account and the ATmega2560 only just happens to work right now.

In this case, it appears that the _board_ is different, with a different upload baudrate. Therefore, a new board definition will be needed. This is therefore not a bug in the arduino-mega2560 board, but lack of support for the arduino-mega1280 target (a missing feature, if you will).

> * Some analog in pins are inverted, and some are duplicated
> 
> * I can write in digital pins but if I try to read, return false no matter what

The Mega 2560 board hasn't seen much testing (actually, the AVR chips in general haven't seen a lot of testing and have known issues). It's very well possible that there are bugs. In which case, if you can find and fix them a PR would be appreciated.

MrNinso, 2021-04-13:
Thanks for merge my PR #1773 , I was thinking to close this issue , but I noticed the atmega2560 don't have a PWM implementation. I'm starting work in this implementation for atmega1280, probably this will work for atmega2560.

did you prefer I rename this issue or open other ? @aykevl @deadprogram

deadprogram, 2021-04-13:
Please open a new issue/PR for that.

I have marked this issue with label for next release, and will close it once it is actually released.

Thank you!

MrNinso, 2021-04-13:
Okay :+1:

deadprogram, 2021-05-12:
This was released with v0.18 so now closing. Thank you!

PR #1773 (2021-04-12): Arduino mega1280 Support
closes #1290

Add Arduino mega 1280 support based in Arduino mega 2560 support.
closed 2021-05-12 by deadprogram