tinygo-org/tinygo #1294: Context package won't compile for arm microcontrollers in v0.14.0

bgould, 2020-08-16:
`tinygo version 0.14.0-dev linux/amd64 (using go version go1.13.3 and LLVM version 10.0.1)`

Given the following program, building for arm microcontroller (`arduino-nano33`, `pyportal`, `feather-nrf52840`, etc):

```
package main

import (
        _ "context"
        "time"
)

func main() {
        for {
                time.Sleep(time.Second)
                println("tick")
        }
}
```

Results in the following compiler error:

```
# context
../../tinygo/src/github.com/tinygo-org/tinygo/src/device/arm/arm.go: interp: unsupported instruction

traceback:
../../tinygo/src/github.com/tinygo-org/tinygo/src/runtime/interrupt/interrupt_cortexm.go:51:22:
  call void @"device/arm.EnableInterrupts"(i32 %state, i8* undef, i8* undef), !dbg !2028
../../tinygo/src/github.com/tinygo-org/tinygo/src/runtime/chan.go:565:19:
  call void @"runtime/interrupt.Restore"(i32 %1, i8* undef, i8* undef), !dbg !2052
../../tinygo/go/src/context/context.go:310:7:
  call void @runtime.chanClose(%runtime.channel* %0, i8* undef, i8* null), !dbg !2024
context/<init>:
  call void @"context.init#1"(i8* undef, i8* undef), !dbg !2031
```

bgould, 2020-08-16:
Also, I confirmed the same problem exists using Go 1.14
`tinygo version 0.14.0-dev linux/amd64 (using go version go1.14.7 and LLVM version 10.0.1)`

niaow, 2020-08-16:
This is #1248

bgould, 2020-08-16:
In this specific case, I believe this is the source of the problem:

https://github.com/golang/go/blob/8c39bbf9c93f773ab351bbddb4c3dd93e4fddc76/src/context/context.go#L339

This code used to work, so I presume that the problem was introduced with the new functionality to be able to wake a goroutine from an interrupt, I guess the disabling of interrupts is like a semaphore or something when a channel gets closed?  When `close` is called on a channel in an initializer, in the context of the `interp` package, it seems like maybe disabling/re-enabled interrupts doesn't actually need to happen?

If that is true, I wonder if something like the following could be possible/feasible:

in (device/arm.go):
```go
// DisableInterrupts disables all interrupts, and returns the old interrupt
// state.
func DisableInterrupts() uintptr {
	if interp.Running() { // returns true only when evaluated by the interp package
		return 42
	}
	return AsmFull(`
		mrs {}, PRIMASK
		cpsid i
	`, nil)
}

// EnableInterrupts enables all interrupts again. The value passed in must be
// the mask returned by DisableInterrupts.
func EnableInterrupts(mask uintptr) {
	if interp.Running() {  // maybe this could be optimized away by the TinyGo compiler when not interpreted?
		return
	}
	AsmFull("msr PRIMASK, {mask}", map[string]interface{}{
		"mask": mask,
	})
}
```

The above suggestion is sort of modeled after the recover() builtin in Go; there are certainly other possibilities, but I think this is relatively easy to understand, gives the programmer a good amount of control for handling cases that can't be interpreted, and maybe could be useful outside of just this specific case.  If you'd like, I can take a stab at implementing something like this following more discussion/feedback on the idea.

niaow, 2020-08-16:
So, this needs to be treated as a synchronization barrier, since the operation may actually wait for an interrupt to fire. So I think treating this as having substantial side-effects is necessary in most cases.

bgould, 2020-08-16:
Oh ok... yes I guess I was figuring that in the context of `interp` (to pre-compute globals during compilation) there aren't actually any interrupts anyhow so enabling/disabling interrupts would not really make a difference.

niaow, 2020-08-16:
> Oh ok... yes I guess I was figuring that in the context of interp (to pre-compute globals during compilation) there aren't actually any interrupts anyhow so enabling/disabling interrupts would not really make a difference.

Well, interp is just running inits, and it is completely possible for an init to interact with hardware. So yeah, it does make a difference because it is a synchronization barrier.

bgould, 2020-09-25:
@aykevl do you have any thoughts about how to be able to get around this issue so that the context package can compile?

aykevl, 2020-10-03:
Yes... the interp package needs to be improved. In fact, I am right now working on a rewrite of the interp package that should avoid most of these issues. It should also fix bugs [like these](https://tinygo.org/lang-support/stdlib/#encoding-asn1):

```
# encoding/asn1
encoding/asn1/: interp: unknown GEP

traceback:
encoding/asn1/:
  %12 = getelementptr inbounds { %"encoding/asn1.BitString" }, { %"encoding/asn1.BitString" }* %11, i32 0, i32 0
```

A function that does something different depending on whether it is running in interp or not (`interp.Running`) seems like a very bad idea to me, it only works around bugs in the interp package. Instead, this interp package should be fixed to be entirely transparent and not cause any errors even if it can't work with some code by falling back to executing an initializer at runtime.

bgould, 2021-10-07:
Closing, this has been fixed and I have confirmed it is working, at least as of v0.19.0