tinygo-org/tinygo #1295: Cant build for arduino on Ubuntu 20.04

bl17zar, 2020-08-16:
System:
```
  Operating System: Ubuntu 20.04 LTS
  Kernel: Linux 5.4.0-42-generic
  Architecture: x86-64
```

tinygo version:
```
version: 0.14.0
```

I followed the instructions on https://tinygo.org/getting-started/linux/ installing all 3 dependecies for AVR. I tried to build simple program for the purpose of checking the resulting file size without the intention to flash it to arduino.

Program:
```go
// filename: hello.go
package main
import "fmt"
func main(){ 
    fmt.Println("Hello World!")
}
```

The error is caused by the following command:
```bash
tinygo build -o hello -target arduino hello.go
```

The error:
```
tinygo: /home/circleci/project/llvm-project/llvm/lib/CodeGen/SelectionDAG/SelectionDAGBuilder.cpp:9794: void llvm::SelectionDAGISel::LowerArguments(const llvm::Function &): Assertion `InVals.size() == Ins.size() && "LowerFormalArguments didn't emit the correct number of values!"' failed.
SIGABRT: abort
PC=0x7f4d2956918b m=5 sigcode=18446744073709551610

goroutine 0 [idle]:
runtime: unknown pc 0x7f4d2956918b
stack: frame={sp:0x7f4d01058540, fp:0x0} stack=[0x7f4d0085a248,0x7f4d01059e48)
00007f4d01058440:  0000000000000006  00000000063bed11
00007f4d01058450:  0000000000002642  00000000063bc1c9
00007f4d01058460:  00000000063bed55  0000000000000004
00007f4d01058470:  0000000000000002  0000000000000000
00007f4d01058480:  0000000000000000  0000000000000000
00007f4d01058490:  0000000000000000  0002ffff00001fa2
00007f4d010584a0:  0000000400000000  0000000000000000
00007f4d010584b0:  0000000000000000  00007f4cec000020
00007f4d010584c0:  00007f4cee3de7d0  0000000000000122
00007f4d010584d0:  0000000000000122  00007f4cee3de7c0
00007f4d010584e0:  0000000000000140  00007f4d295c12d6
00007f4d010584f0:  3439373901058550  79fb17c5a0b0c700
00007f4d01058500:  00007f4c7ca21a80  00007f4d010587a0
00007f4d01058510:  0000000000000121  0000000000000121
00007f4d01058520:  0000000000000122  00007f4cee3de7d0
00007f4d01058530:  00007f4d01058550  00007f4d295b1d89
00007f4d01058540: <0000000000000000  00007f4d295c0850
00007f4d01058550:  00000404fbad8000  00007f4cee3de7d0
00007f4d01058560:  00007f4cee3de835  00007f4cee3de7d0
00007f4d01058570:  00007f4cee3de7d0  00007f4cee3de8f1
00007f4d01058580:  00007f4cee3de8fc  00007f4cee3de7d0
00007f4d01058590:  00007f4cee3de8fc  0000000000000000
00007f4d010585a0:  0000000000000000  0000000000000000
00007f4d010585b0:  0000000000000000  0000000000000000
00007f4d010585c0:  fffffffe7fffffff  ffffffffffffffff
00007f4d010585d0:  ffffffffffffffff  ffffffffffffffff
00007f4d010585e0:  ffffffffffffffff  ffffffffffffffff
00007f4d010585f0:  ffffffffffffffff  ffffffffffffffff
00007f4d01058600:  ffffffffffffffff  ffffffffffffffff
00007f4d01058610:  ffffffffffffffff  ffffffffffffffff
00007f4d01058620:  ffffffffffffffff  ffffffffffffffff
00007f4d01058630:  ffffffffffffffff  ffffffffffffffff
runtime: unknown pc 0x7f4d2956918b
stack: frame={sp:0x7f4d01058540, fp:0x0} stack=[0x7f4d0085a248,0x7f4d01059e48)
00007f4d01058440:  0000000000000006  00000000063bed11
00007f4d01058450:  0000000000002642  00000000063bc1c9
00007f4d01058460:  00000000063bed55  0000000000000004
00007f4d01058470:  0000000000000002  0000000000000000
00007f4d01058480:  0000000000000000  0000000000000000
00007f4d01058490:  0000000000000000  0002ffff00001fa2
00007f4d010584a0:  0000000400000000  0000000000000000
00007f4d010584b0:  0000000000000000  00007f4cec000020
00007f4d010584c0:  00007f4cee3de7d0  0000000000000122
00007f4d010584d0:  0000000000000122  00007f4cee3de7c0
00007f4d010584e0:  0000000000000140  00007f4d295c12d6
00007f4d010584f0:  3439373901058550  79fb17c5a0b0c700
00007f4d01058500:  00007f4c7ca21a80  00007f4d010587a0
00007f4d01058510:  0000000000000121  0000000000000121
00007f4d01058520:  0000000000000122  00007f4cee3de7d0
00007f4d01058530:  00007f4d01058550  00007f4d295b1d89
00007f4d01058540: <0000000000000000  00007f4d295c0850
00007f4d01058550:  00000404fbad8000  00007f4cee3de7d0
00007f4d01058560:  00007f4cee3de835  00007f4cee3de7d0
00007f4d01058570:  00007f4cee3de7d0  00007f4cee3de8f1
00007f4d01058580:  00007f4cee3de8fc  00007f4cee3de7d0
00007f4d01058590:  00007f4cee3de8fc  0000000000000000
00007f4d010585a0:  0000000000000000  0000000000000000
00007f4d010585b0:  0000000000000000  0000000000000000
00007f4d010585c0:  fffffffe7fffffff  ffffffffffffffff
00007f4d010585d0:  ffffffffffffffff  ffffffffffffffff
00007f4d010585e0:  ffffffffffffffff  ffffffffffffffff
00007f4d010585f0:  ffffffffffffffff  ffffffffffffffff
00007f4d01058600:  ffffffffffffffff  ffffffffffffffff
00007f4d01058610:  ffffffffffffffff  ffffffffffffffff
00007f4d01058620:  ffffffffffffffff  ffffffffffffffff
00007f4d01058630:  ffffffffffffffff  ffffffffffffffff

goroutine 1 [syscall]:
runtime.cgocall(0xd80fa0, 0xc00155f5a8, 0xc00635aaf8)
        /usr/local/go/src/runtime/cgocall.go:133 +0x5b fp=0xc00155f578 sp=0xc00155f540 pc=0x98604b
	tinygo.org/x/go-llvm._Cfunc_LLVMTargetMachineEmitToMemoryBuffer(0x7ce6220, 0x7d03990, 0x1, 0xc00635aaf0, 0xc00635aaf8, 0xc000000000)
	        _cgo_gotypes.go:9409 +0x4d fp=0xc00155f5a8 sp=0xc00155f578 pc=0xc284bd
		tinygo.org/x/go-llvm.TargetMachine.EmitToMemoryBuffer.func1(0x7ce6220, 0x7d03990, 0xc000000001, 0xc00635aaf0, 0xc00635aaf8, 0x1b)
		        /go/pkg/mod/tinygo.org/x/go-llvm@v0.0.0-20200503225853-345b2947b59d/target.go:273 +0x111 fp=0xc00155f5e8 sp=0xc00155f5a8 pc=0xc51391
			tinygo.org/x/go-llvm.TargetMachine.EmitToMemoryBuffer(0x7ce6220, 0x7d03990, 0x1, 0xc00636ce20, 0x1b, 0x14)
			        /go/pkg/mod/tinygo.org/x/go-llvm@v0.0.0-20200503225853-345b2947b59d/target.go:273 +0x84 fp=0xc00155f650 sp=0xc00155f5e8 pc=0xc376a4
				github.com/tinygo-org/tinygo/builder.Build(0x7fff42dd018d, 0x8, 0x7fff42dd0177, 0x5, 0xc00024c4e0, 0xc00155fc88, 0x0, 0x0)
				        /home/circleci/project/builder/build.go:145 +0x628 fp=0xc00155fc48 sp=0xc00155f650 pc=0xd4c088
					main.Build(0x7fff42dd018d, 0x8, 0x7fff42dd0177, 0x5, 0xc00013e0e0, 0x0, 0x15)
					        /home/circleci/project/main.go:97 +0xc5 fp=0xc00155fcb0 sp=0xc00155fc48 pc=0xd5c835
						main.main()
						        /home/circleci/project/main.go:888 +0x1b7f fp=0xc00155ff88 sp=0xc00155fcb0 pc=0xd6144f
							runtime.main()
							        /usr/local/go/src/runtime/proc.go:203 +0x1fa fp=0xc00155ffe0 sp=0xc00155ff88 pc=0x9b84aa
								runtime.goexit()
								        /usr/local/go/src/runtime/asm_amd64.s:1373 +0x1 fp=0xc00155ffe8 sp=0xc00155ffe0 pc=0x9e5381

									rax    0x0
									rbx    0x7f4d0105a700
									rcx    0x7f4d2956918b
									rdx    0x0
									rdi    0x2
									rsi    0x7f4d01058540
									rbp    0x7f4d296de588
									rsp    0x7f4d01058540
									r8     0x0
									r9     0x7f4d01058540
									r10    0x8
									r11    0x246
									r12    0x63bc1c9
									r13    0x2642
									r14    0x63bed55
									r15    0x7f4c7d90f520
									rip    0x7f4d2956918b
									rflags 0x246
									cs     0x33
									fs     0x0
									gs     0x0
```

deadprogram, 2020-08-17:
Please see https://github.com/tinygo-org/tinygo/issues/1296#issuecomment-674554505

bl17zar, 2020-08-18:
@deadprogram it really helped. Thank you!

closed 2020-08-18 by bl17zar