tinygo-org/tinygo #1296: Arduino compile bug

ncw, 2020-08-16:
Try compiling the following program

```go
package main

import (
	"fmt"
)

func main() {
	x := uint16(0x4242)
	s := fmt.Sprintf("%d", x)
	println(s)
}
```

With

```
tinygo build -o /tmp/z -target arduino .
```

The result is

```
tinygo build -o /tmp/z -target arduino .
tinygo: /home/circleci/project/llvm-project/llvm/lib/CodeGen/SelectionDAG/SelectionDAGBuilder.cpp:9794: void llvm::SelectionDAGISel::LowerArguments(const llvm::Function &): Assertion `InVals.size() == Ins.size() && "LowerFormalArguments didn't emit the correct number of values!"' failed.
SIGABRT: abort
PC=0x7f8cab99518b m=7 sigcode=18446744073709551610

goroutine 0 [idle]:
runtime: unknown pc 0x7f8cab99518b
stack: frame={sp:0x7f8c7e34a540, fp:0x0} stack=[0x7f8c7d9bd248,0x7f8c7e34be48)
00007f8c7e34a440:  0000000000000006  00000000063bed11 
00007f8c7e34a450:  0000000000002642  00000000063bc1c9 
00007f8c7e34a460:  00000000063bed55  0000000000000000 
00007f8c7e34a470:  0000000000000002  0000000000000000 
00007f8c7e34a480:  0000000000000000  0000000000000000 
00007f8c7e34a490:  0000000000000000  0000ffff00001fa2 
00007f8c7e34a4a0:  0000000400000000  0000000000000000 
00007f8c7e34a4b0:  0000000000000000  00007f8c68000020 
00007f8c7e34a4c0:  00007f8c68248cb0  0000000000000122 
00007f8c7e34a4d0:  0000000000000122  00007f8c68248ca0 
00007f8c7e34a4e0:  0000000000000140  00007f8cab9ed2d6 
00007f8c7e34a4f0:  343937397e34a550  7210b33d8f611300 
00007f8c7e34a500:  00007f8c6b86b680  00007f8c7e34a7a0 
00007f8c7e34a510:  0000000000000121  0000000000000121 
00007f8c7e34a520:  0000000000000122  00007f8c68248cb0 
00007f8c7e34a530:  00007f8c7e34a550  00007f8cab9ddd89 
00007f8c7e34a540: <0000000000000000  00007f8cab9ec850 
00007f8c7e34a550:  00000404fbad8000  00007f8c68248cb0 
00007f8c7e34a560:  00007f8c68248d15  00007f8c68248cb0 
00007f8c7e34a570:  00007f8c68248cb0  00007f8c68248dd1 
00007f8c7e34a580:  00007f8c68248ddc  00007f8c68248cb0 
00007f8c7e34a590:  00007f8c68248ddc  0000000000000000 
00007f8c7e34a5a0:  0000000000000000  0000000000000000 
00007f8c7e34a5b0:  0000000000000000  0000000000000000 
00007f8c7e34a5c0:  fffffffe7fffffff  ffffffffffffffff 
00007f8c7e34a5d0:  ffffffffffffffff  ffffffffffffffff 
00007f8c7e34a5e0:  ffffffffffffffff  ffffffffffffffff 
00007f8c7e34a5f0:  ffffffffffffffff  ffffffffffffffff 
00007f8c7e34a600:  ffffffffffffffff  ffffffffffffffff 
00007f8c7e34a610:  ffffffffffffffff  ffffffffffffffff 
00007f8c7e34a620:  ffffffffffffffff  ffffffffffffffff 
00007f8c7e34a630:  ffffffffffffffff  ffffffffffffffff 
runtime: unknown pc 0x7f8cab99518b
stack: frame={sp:0x7f8c7e34a540, fp:0x0} stack=[0x7f8c7d9bd248,0x7f8c7e34be48)
00007f8c7e34a440:  0000000000000006  00000000063bed11 
00007f8c7e34a450:  0000000000002642  00000000063bc1c9 
00007f8c7e34a460:  00000000063bed55  0000000000000000 
00007f8c7e34a470:  0000000000000002  0000000000000000 
00007f8c7e34a480:  0000000000000000  0000000000000000 
00007f8c7e34a490:  0000000000000000  0000ffff00001fa2 
00007f8c7e34a4a0:  0000000400000000  0000000000000000 
00007f8c7e34a4b0:  0000000000000000  00007f8c68000020 
00007f8c7e34a4c0:  00007f8c68248cb0  0000000000000122 
00007f8c7e34a4d0:  0000000000000122  00007f8c68248ca0 
00007f8c7e34a4e0:  0000000000000140  00007f8cab9ed2d6 
00007f8c7e34a4f0:  343937397e34a550  7210b33d8f611300 
00007f8c7e34a500:  00007f8c6b86b680  00007f8c7e34a7a0 
00007f8c7e34a510:  0000000000000121  0000000000000121 
00007f8c7e34a520:  0000000000000122  00007f8c68248cb0 
00007f8c7e34a530:  00007f8c7e34a550  00007f8cab9ddd89 
00007f8c7e34a540: <0000000000000000  00007f8cab9ec850 
00007f8c7e34a550:  00000404fbad8000  00007f8c68248cb0 
00007f8c7e34a560:  00007f8c68248d15  00007f8c68248cb0 
00007f8c7e34a570:  00007f8c68248cb0  00007f8c68248dd1 
00007f8c7e34a580:  00007f8c68248ddc  00007f8c68248cb0 
00007f8c7e34a590:  00007f8c68248ddc  0000000000000000 
00007f8c7e34a5a0:  0000000000000000  0000000000000000 
00007f8c7e34a5b0:  0000000000000000  0000000000000000 
00007f8c7e34a5c0:  fffffffe7fffffff  ffffffffffffffff 
00007f8c7e34a5d0:  ffffffffffffffff  ffffffffffffffff 
00007f8c7e34a5e0:  ffffffffffffffff  ffffffffffffffff 
00007f8c7e34a5f0:  ffffffffffffffff  ffffffffffffffff 
00007f8c7e34a600:  ffffffffffffffff  ffffffffffffffff 
00007f8c7e34a610:  ffffffffffffffff  ffffffffffffffff 
00007f8c7e34a620:  ffffffffffffffff  ffffffffffffffff 
00007f8c7e34a630:  ffffffffffffffff  ffffffffffffffff 

goroutine 1 [syscall]:
runtime.cgocall(0xd80fa0, 0xc006b8f5a8, 0xc0036df630)
	/usr/local/go/src/runtime/cgocall.go:133 +0x5b fp=0xc006b8f578 sp=0xc006b8f540 pc=0x98604b
tinygo.org/x/go-llvm._Cfunc_LLVMTargetMachineEmitToMemoryBuffer(0x93d0220, 0x93ed9a0, 0x1, 0xc0036df628, 0xc0036df630, 0xc000000000)
	_cgo_gotypes.go:9409 +0x4d fp=0xc006b8f5a8 sp=0xc006b8f578 pc=0xc284bd
tinygo.org/x/go-llvm.TargetMachine.EmitToMemoryBuffer.func1(0x93d0220, 0x93ed9a0, 0xc000000001, 0xc0036df628, 0xc0036df630, 0x1b)
	/go/pkg/mod/tinygo.org/x/go-llvm@v0.0.0-20200503225853-345b2947b59d/target.go:273 +0x111 fp=0xc006b8f5e8 sp=0xc006b8f5a8 pc=0xc51391
tinygo.org/x/go-llvm.TargetMachine.EmitToMemoryBuffer(0x93d0220, 0x93ed9a0, 0x1, 0xc003726460, 0x1b, 0x14)
	/go/pkg/mod/tinygo.org/x/go-llvm@v0.0.0-20200503225853-345b2947b59d/target.go:273 +0x84 fp=0xc006b8f650 sp=0xc006b8f5e8 pc=0xc376a4
github.com/tinygo-org/tinygo/builder.Build(0x7ffea23a0261, 0x1, 0x7ffea23a024a, 0x6, 0xc00022a600, 0xc006b8fc88, 0x0, 0x0)
	/home/circleci/project/builder/build.go:145 +0x628 fp=0xc006b8fc48 sp=0xc006b8f650 pc=0xd4c088
main.Build(0x7ffea23a0261, 0x1, 0x7ffea23a024a, 0x6, 0xc0001b00e0, 0x0, 0x15)
	/home/circleci/project/main.go:97 +0xc5 fp=0xc006b8fcb0 sp=0xc006b8fc48 pc=0xd5c835
main.main()
	/home/circleci/project/main.go:888 +0x1b7f fp=0xc006b8ff88 sp=0xc006b8fcb0 pc=0xd6144f
runtime.main()
	/usr/local/go/src/runtime/proc.go:203 +0x1fa fp=0xc006b8ffe0 sp=0xc006b8ff88 pc=0x9b84aa
runtime.goexit()
	/usr/local/go/src/runtime/asm_amd64.s:1373 +0x1 fp=0xc006b8ffe8 sp=0xc006b8ffe0 pc=0x9e5381

rax    0x0
rbx    0x7f8c7e34c700
rcx    0x7f8cab99518b
rdx    0x0
rdi    0x2
rsi    0x7f8c7e34a540
rbp    0x7f8cabb0a588
rsp    0x7f8c7e34a540
r8     0x0
r9     0x7f8c7e34a540
r10    0x8
r11    0x246
r12    0x63bc1c9
r13    0x2642
r14    0x63bed55
r15    0x7f8c61844080
rip    0x7f8cab99518b
rflags 0x246
cs     0x33
fs     0x0
gs     0x0

Compilation exited abnormally with code 2 at Sat Aug 15 18:08:27
```

Tested with

```
$ tinygo version
tinygo version 0.14.0 linux/amd64 (using go version go1.14.3 and LLVM version 10.0.1)
```

aykevl, 2020-08-16:
Yes, this is a known issue in the AVR backend of LLVM (not directly in TinyGo). You should avoid importing the fmt package on Arduino. Even if it worked, it would probably not fit on the small flash of the chip. Instead, you can use plain `println` or write directly to the UART.

deadprogram, 2020-09-03:
Closing as duplicate. Please reopen if needed. Thanks!

closed 2020-09-03 by deadprogram

ncw, 2020-09-04:
No problems! I'm finding the AVR backend to be extremely buggy, I guess this is normal?

Thanks

aykevl, 2020-09-04:
Yes, it is still quite buggy. This is much improved in LLVM 11, but I haven't yet fixed this particular bug (so you can't use the fmt package yet - and you shouldn't, it's way too big for an AVR).