tinygo-org/tinygo #1314: Question re non bare metal

mcrosson, 2020-08-25:
I know the TinyGo project is focused on MCU support and native binaries for them. However, some arm boards (raspberry pi, friendly arm boards) have native i2c, gpio, spi, etc while also running Linux. Is there a way for those boards to be supported by TinyGo while remaining inside the scope of the project, or is supporting Linux based boxes with various forms of peripheries secondary and/or outside the scope of the project?

This is a pure question, I'm not asking for a change in project scope.

aykevl, 2020-08-26:
Right now it's not really in scope, because the main Go toolchain already supports these ARM boards. For example, you should be able to use the github.com/google/periph package on some of these boards already. TinyGo is mainly focused on systems that the main Go toolchain can't target right now because of binary size (among other reasons).
In other words, it's not explicitly out of scope, but it's also not in scope because there should be a working alternative.

Is there a reason why you'd like to compile with TinyGo on these ARM boards?

That said, TinyGo does in fact support most Linux (and macOS) systems. For example, all unit tests pass on supported host systems. You can either compile a program directly on one of these systems, or use `GOARCH=arm` (or similar) to target these systems.

mcrosson, 2020-08-26:
>  In other words, it's not explicitly out of scope, but it's also not in scope because there should be a working alternative.

Understood. I thought that was the case but I wanted to be sure.

> Is there a reason why you'd like to compile with TinyGo on these ARM boards?

Driver support. Most of the alternatives can be used, yes, but I would need to write drivers for any hardware I would like to use.  I was hoping to leverage the strong driver support TinyGo has in those places. Fortunately I am working on an open project so I could port drivers from TinyGo as needed.

> That said, TinyGo does in fact support most Linux (and macOS) systems

I had noticed the support for building natively on Linux systems. I started with this fact and quickly realized I2C and SPI were not setup for a non-machine build. Unfortunately I would need i2c/spi for this project and can't use TinyGo as-is. This is *not* a problem given the scope of TinyGO.

closed 2020-08-26 by mcrosson