tinygo-org/tinygo #1320: There is a problem after writing atsamd with openocd

sago35, 2020-08-26:
#1310 makes it easier to write from openocd with feather-m4 and others.
However, two problems remain.

### 1. touchSerialPortAt1200bps() is executed when programmer is specified

As for 1, the following json does not have the flash-1200-bps-reset setting and will not cause a problem.
However, it is not reasonable to separate json, so you need to set flash-1200-bps-reset = false when -programmer is specified.

```
{
    "inherits": ["atsamd51j19a"],
    "build-tags": ["sam", "atsamd51j19a", "feather_m4"]
}
```

If you solve this problem, we will be able to write to the board in panic state.
(Normally flash-1200-bps-reset cannot be performed on a board in panic state.)

Also, we will be able to do rewrites very quickly.

### 2. Sometimes bootloader mode is triggered after writing

I think the reason for this problem is that the reset is performed multiple times when writing.
I don't know why, but changing `program *.hex reset exit` to `program *.hex verify reset exit` solved the problem.
I don't know if this is the right solution, but I'd like to fix it in some way.

sago35, 2020-08-26:
Both issues 1 and 2 also occurred with feather-nrc52840.

sago35, 2020-09-03:
> 1. touchSerialPortAt1200bps() is executed when programmer is specified

fixed. (flash: call PortReset only on other than openocd #1335)

deadprogram, 2020-09-18:
This issue was corrected in v0.15.0 which was just released. Now closing.

sago35, 2020-09-18:
> 2. Sometimes bootloader mode is triggered after writing
This one is still unresolved.

If I check with feather-m4 and wioterminal now, it's almost certainly a bootloader state.
The writes are successful, so when I reset it, it starts working normally.

Trying it with feather-nrf52840 and xiao did not reproduce the problem.

* environment
    * ubuntu 20.04
    * jlink edu mini
    * cmsis-dap-v2 debug adptor (XIAO)

deadprogram, 2021-04-22:
@sago35 is this still an issue?

sago35, 2021-04-22:
> Sometimes bootloader mode is triggered after writing
This one is still unresolved.

@deadprogram 

For the adafruit board, updating the bootloader fixed the problem.
However, I still have the following problem with wioterminal.

I would like #1397 to be merged.

However, at this point, I still don't know why adding verify eliminates the problem.
This is a problem with openocd, and the same problem occurs when I run openocd directly without tinygo.

niaow, 2022-11-04:
Is this still an issue now that #1397 is merged?

deadprogram, 2024-11-30:
Closing since this appears to have been handled. Thanks everyone!

closed 2024-11-30 by deadprogram